terjeio/ioSender #243: Work coodinates reset at the program end

calabr, 2022-08-26:
sender 2.0.40p2
system GRBL 1.1h
 selecting work offset before ptogram start (G54 ) will use this offset as current working in the program.
 After program finished, iosender reset current offset to G53, but still show G54 selected in Work Parameters->Offset
 next start of the same program will use G53 as default.
  sender should not do any configuration without command.
 My use-case is mesure several workpiecees, remember offsets, then set cutter and run the same program for different workpieces.

terjeio, 2022-08-26:
G53 is not an offset - it is for [motion in machine coordinates](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g53) and is valid only for the line it is present in.
At program end (M2 or M30) the active coordinate system is set to G54, this is [normal behaviour](https://www.linuxcnc.org/docs/html/gcode/m-code.html#mcode:m2-m30).

calabr, 2022-08-26:
Sorry, I mean manually set G55, start program in G55, sender will reset it to G54 after programm ended. 
  It's 2 confusing behawiour there -
  1. machine working coordinates after program completed differ to work coordinate after program finished, but program didn't change it
  2. sender changed something, but don't reflect it in GUI.
 Logically I can see 2 more fail-safe scenario:
  1) sender use current CS as default one and return to it after M30 M2 - preferablr
  2) sender didn't change anything and track, reflect in the GUI current state.
3) - allow both and give the user select it in config file.

terjeio, 2022-08-26:
It is the [controller that resets the coordinate system to G54](https://github.com/grblHAL/core/blob/5dd9c3eb18b6a0757de65d3fa49912882489355b/gcode.c#L3043-L3107) on program end, not the sender. See line 3062 and lines 3081 in the linked code segment. This is normal behaviour - grblHAL, Grbl and LinuxCNC does it and I guess most other controllers as it is part of the NIST RS274NGC v3 specification ([see page 38](https://duckduckgo.com/?t=ffsb&q=NIST+RS274NGC+v3&ia=web)).

calabr, 2022-09-16:
Thank you to point me right direction!
 You right. GRPL resets to G54 after M30 and M2. 
Just ioSender didn't change current coordinate system in the GUI - probable this issue is inly with legacy GRBL. I believe grblHAL reports proactively of internal state change. 
 is it possible to request state after program end and synchronize sender?

terjeio, 2022-09-16:
>  I believe grblHAL reports proactively of internal state change.

Not always, I have missed program end if _Parser state_  is selected set for `$10` - status report options. Will fix in next commit.

> is it possible to request state after program end and synchronize sender?

I do so already, here is the console output (with _Verbose_ checked) from Grbl 1.1f on a Mega:

```
g55
<Idle|WPos:-2.000,0.000,0.000|FS:0,0|WCO:2.000,0.000,0.000>
<Idle|WPos:-2.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Idle|WPos:-2.000,0.000,0.000|FS:0,0>
m30
[MSG:Pgm End]
[GC:G1 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
<Idle|WPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
<Idle|WPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Idle|WPos:0.000,0.000,0.000|FS:0,0>
```

You do not get the parser state response?

calabr, 2022-09-24:
If program runs automatically - I see
`<Run|MPos:-117.280,-106.488,-51.828|Bf:14,128|FS:4000,0>
<Run|MPos:-117.280,-106.488,-44.623|Bf:14,128|FS:1165,0>
[MSG:Pgm End]
<Idle|MPos:-117.280,-106.488,-42.895|Bf:15,128|FS:0,0|WCO:-191.208,-116.488,-62.895>
<Idle|MPos:-117.280,-106.488,-42.895|Bf:15,128|FS:0,0|Ov:100,100,100>
<Idle|MPos:-117.280,-106.488,-42.895|Bf:15,128|FS:0,0>
`
If I manually run 
G55
M30
`G55
<Idle|MPos:-117.280,-106.488,-42.895|Bf:15,128|FS:0,0|WCO:-127.280,-116.488,-62.895>
M30
[MSG:Pgm End]
[GC:G1 G54 G17 G21 G90 G94 M5 M9 M56 T0 F1000 S0]
<Idle|MPos:-117.280,-106.488,-42.895|Bf:15,128|FS:0,0|WCO:-191.208,-116.488,-62.895>`
But Offset list-box still show G56, which I set before starting program.

terjeio, 2022-09-25:
I have fixed this in the latest commit of grblHAL. Legacy Grbl should be made a PR for with the neccesary changes?

terjeio, 2022-09-25:
I have looked at changing the sender for lack of feedback from legacy Grbl, but I give up for now as I have to wait for coordinate changes having been executed in the controller before I can request a parser state request. Not easy to implement...

IMO somebody should get access to the Grbl repo so that PR requests could be merged - it has been dead for a long time now and if nobody takes over it is time to move on?

BTW, as a workaround in ioSender you may create a macro for `$G` and run it manually.

closed 2022-09-25 by terjeio