terjeio/ioSender #246: CollectionView does not support changes to its source collection

cfficaurzua, 2022-09-01:
[image]

I got this error when clicking on grbl settings.

terjeio, 2022-09-03:
Works for me. Do you have any additional information that might be helpful?
Windows OS, grbl type and build date (from $I), sender version...

cfficaurzua, 2022-09-05:
Hi i am connecting via Network (TELNET) to an Ortur Aufero CNC
After successfully connecting this logs into the console.
```
Ortur Aufero Ready!
OCF 103.
Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
[echo: $I+]
error:3
[echo: $I+]
error:3
[echo: $I+]
error:3
[echo: $I+]
error:3
[echo: $I+]
error:3
```
After command $I
```
[echo: $I]
[ORIGIN: China]
[PRODUCER: ORTUR]
[AUTHOR: RenShen]
[MODEL: Ortur Aufero]
[OCF: 103]
[DATE:18:20:48 - Apr 16 2021]
[VER:1.1f(STM32F103CB).20201103:]
[OPT:VNMZHSL+W,31,512,4]
```

[image]

I am using Windows 11 Home Single Language Version 10.0.22000 Build 22000

When I clicked in the  Settings: GRBL tab, the error happens

Thanks

cfficaurzua, 2022-09-07:
I got this output

```
$$
[echo: $$]
$0=2.0
$1=5000
$2=0
$3=2
$4=0
$5=7
$6=1
$10=255
$11=0.010
$12=0.002
$13=0
$14=8
$15=0
$16=1
$17=0
$18=0
$19=0
$20=1
$21=1
$22=43
$23=3
$24=60.000
$25=600.000
$26=250
$27=0.500
$28=0.100
$29=0.0
$30=10000.
$31=0.
$32=0
$33=1000.000
$34=0.000
$35=0.000
$36=100.000
$37=0
$39=1
$40=1
$41=0
$42=2
$43=1
$44=4
$45=3
$46=0
$47=0
$56=0.000
$57=100.000
$58=0.000
$59=500.000
$60=0
$61=0
$62=0
$63=2
$64=1
$65=0
$100=800.000
$101=800.000
$102=800.000
$103=800.000
$110=1200.000
$111=1200.000
$112=1200.000
$113=6000.000
$120=10.000
$121=10.000
$122=10.000
$123=10.000
$130=300.000
$131=180.000
$132=50.000
$133=0.000
ok
<Alarm|MPos:0.000,0.000,0.000,0.000|Bf:31,512|FS:0,0|Pn:HS>

<Alarm|MPos:0.000,0.000,0.000,0.000|Bf:31,512|FS:0,0|Pn:HS>
```

terjeio, 2022-09-15:
I have modified a STM32F756 controller to output the mangled startup and $I responses and I can still connect, both via USB and telnet. The grbl settings tab is messed up but that is as expected.
The exception could be be due to something else but since I am not able to replicate it I am at a loss about what to try next.

cfficaurzua, 2022-09-15:
Is there anything I can do to debug ? It only fails in the telnet mode.
I tried out the isLegacy mode, and it works like a charm
Thanks

terjeio, 2022-09-18:
> I tried out the isLegacy mode, and it works like a charm

In telnet mode as well? If so that could be the solution as the responses coming from the controller is bound to confuse the sender.

closed 2023-04-29 by terjeio