terjeio/ioSender #247: program close itself after error

drom89, 2022-09-29:
If i got any error after connecting to ESP32 (MKS DLC32 V2.1), the app just close it self...
So i cant tinker in it :(, is it possible to some how launch it in some kind of debug mode ? 

[image]

The main reason for errors is pretty much cuz i got badly compiled firmware :( , but that is another story :) 

[VER:1.1f.20220928:]
[OPT:VNMSL,35,1024,3,0]
Target buffer size found
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,REBOOT,TC,SED,RTC]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:ESP32]
[DRIVER VERSION:220922]
[DRIVER OPTIONS:v4.4.2]
[BOARD:MKS DLC32 2.0]

LoganFraser, 2022-09-30:
IOSender is designed to read all the parameters set in the firmware, that's how it functions.  If it can't connect to the controller, then it can't do anything anyway.

terjeio, 2022-09-30:
Odd, this is likely due to corrupted communication with the controller.

> So i cant tinker in it :(, is it possible to some how launch it in some kind of debug mode ?

No, not with the release builds.

drom89, 2022-09-30:
So i recompiled with VScode and ESP-IDF and i still getting errors from iosender (iosender close itself after OK button).
If i connect thru Lightburn i at least get some info. But i do not have connected motors, so i dont know if it is working.
[image]
[image]
[image]

terjeio, 2022-09-30:
Same errors with the latest edge version?

drom89, 2022-09-30:
Its weird, every time i start the program i get different errors.

[image]
[image]

After few restarts i managed to get in the console.

```
[DRIVER:ESP32]
[DRIVER VERSION:220922]
[DRIVER OPTIONS:v4.4.2]
[BOARD:MKS DLC32 2.0]
[G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]
[G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]
error:2 - Bad number format: Missing the expected G-code word value or numeric value format is not valid.
[G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]
error:2 - Bad number format: Missing the expected G-code word value or numeric value format is not valid.
[G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]
[... 114 lines cut ...]
error:2 - Bad number format: Missing the expected G-code word value or numeric value format is not valid.
error:2 - Bad number format: Missing the expected G-code word value or numeric value format is not valid.
error:2 - Bad number format: Missing the expected G-code word value or numeric value format is not valid.
error:2 - Bad number format: Missing the expected G-code word value or numeric value format is not valid.
error:2 - Bad number format: Missing the expected G-code word value or numeric value format is not valid.
error:2 - Bad number format: Missing the expected G-code word value or numeric value format is not valid.
error:2 - Bad number format: Missing the expected G-code word value or numeric value format is not valid.
error:2 - Bad number format: Missing the expected G-code word value or numeric value format is not valid.
error:2 - Bad number format: Missing the expected G-code word value or numeric value format is not valid.
error:2 - Bad number format: Missing the expected G-code word value or numeric value format is not valid.
ok
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ|Ov:80,50,100>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ|WCO:0.000,0.000,0.000>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ|Ov:80,50,100>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ>
```

drom89, 2022-09-30:
Any way the main problem was in the firmware. The program closing itself after some error is minor issue...

closed 2022-09-30 by drom89