terjeio/ioSender #248: Feature Request: Hard Limits on Z only.

LoganFraser, 2022-09-30:
Many 4 wire toolsetters have two sets of NC switches - for set and overtravel.  

I would like to be able to use Overtravel in series with a Zmax endstop to save inputs - basically having hard limits enabled on the Z axis only with toolsetter over-travel as "z-min" essentially.   

Is this a possible addition?  Thanks.

terjeio, 2022-09-30:
Yes? Not sure what the addition should be...

For grblHAL based controllers: what could be done is to write a small plugin that redirects Z-limit interrupts to the _probe disconnected_ signal when probing, this would then trigger a feed hold.

LoganFraser, 2022-09-30:
I think it'd be useful to have it on *all* the time as opposed to just when probing, in case of pathing issues where the toolsetter is driven down on. Basically just acting like a hard hard limit to prevent tool or setter damage.

terjeio, 2022-10-02:
if it is acceptable to do a feed hold instead of triggering a hard limit alarm then a plugin would be better? The position is likely lost when a hard limit alarm is raised, it is not if the program is stopped after a feed hold.

Per axis enabling of hard limits will require a new core setting, `$21` cannot be modified without breaking backwards compatability...

terjeio, 2023-04-29:
Closing this as it cannot be implemented sender side.
A grblHAL plugin or board code, [similar to this](https://github.com/grblHAL/ESP32/blob/master/main/BlackBoxX32.c), can modify hard limits handling - a way to implement this for now?

closed 2023-04-29 by terjeio