terjeio/ioSender #251: Touch Probe Material Measuring

Xchrisfox, 2022-10-27:
I'm looking for a solution to use the touch probe for material measuring.

Right now it seems that when using the touch probe it automatically zeroes the WCS.

Wanting to use this for measuring parts.

calabr, 2022-11-04:
I'm also want's to join to this feature request.
it's will be useful to make measurements, see the results, and than decide to apply it or not.

terjeio, 2022-11-09:
How should this work? Zero at an anchor position and measure relative to that? And how/where to copy the result? From the clipboard (in which format?) and/or the console? Or the DRO?

calabr, 2022-11-11:
Setup probing as for regular action, check checkbox "Measure only". 
Start probing as always, and show result on the same page with probe - Just as it's appear from the probe and, re-calculated to current working system. 
If applicable - it's will be good to show additional calculation, like angle, distance from current point, max size for internal and external center finder etc. 
Than activate button apply. - use measurement result for working system zero etc. 
If error received - big readable error message instead of results with error details.
Iven if we choose to apply results right after probing - will be good to fill-in results on probing page.
 Always visible DRO can also help for better use, but it's different story.

terjeio, 2022-11-12:
Initial changes for measure has been [uploaded](http://www.io-engineering.com/downloads). Measured coordinate is written to the console.

> If applicable - it's will be good to show additional calculation, like angle, distance from current point, max size for internal and external center finder etc.

Applicable for what purpose? I assume measurments are for transferring into an external app. Am I wrong about that?
Some use case scenarios would help my understanding.

> Iven if we choose to apply results right after probing...

I'll leave the original functionality for setting a coordinate system offset or G92 offset. Were you thinking of something else?

> ... - will be good to fill-in results on probing page.

By repurposing the preview frame?

> Always visible DRO can also help for better use, but it's different story.

I can add that for the XL version - not enough UI real estate in the standard version (unless visibility is made dynamic).

calabr, 2022-11-14:
> Applicable for what purpose? I assume measurments are for transferring into an external app. Am I wrong about that?
> Some use case scenarios would help my understanding.

I mean if we can calculate something helpful from results, like rotation angle,  displacement from current zero, X and Y size.
Remember current point, or current zero candidate, and draw it on 3D view. ( for example we can check if hall in the material not crossed by cutting path, or where fixture located and it's height. 
For example, I want to place workpiece striate over axis - measure angle, but angle itself didn't give much info I better need step of first point to second. and I will several measurements with with workpiece move to satisfy. 

> I'll leave the original functionality for setting a coordinate system offset or G92 offset. Were you thinking of something else?

Current functionality dictate strict workflow - select offset/WS, prepare probe and WS will be zeroed.
Apply after measure give flexibility - you can decide which WS to set, or move blank and measure again.

> By repurposing the preview frame?

Yes. Or space, occupied by it. Probe result is also strictly needed on probe page, not in status bar. I found disappointed every time after probing finished, especially in case of error.

> I can add that for the XL version - not enough UI real estate in the standard version (unless visibility is made dynamic).

It's can be a good start.
I can propose you how to rearrange view to fit DRO. Some buttons should also be moved on GRBL page.
I tried to do it myself, but not enough of knowledge in C#

terjeio, 2022-11-14:
DRO has been added in the [latest edge version](http://www.io-engineering.com/downloads), with dynamic visibility for the standard version. I need to work out how to approach the rest, both how to accumulate measurements and how to present and interact with them. Currently they are still written to the console, a detached view can be opened via the _File_ menu.

calabr, 2022-11-15:
I'm tryed XL version - DRO itself really help's to understand what is going on during proping. It's give visual feedback on moving and zeroing coordinates.
My CNC PC is 640x480, so it's cut right part of the screen, and all probbing tabs fit's well. The only height map frame not fit, but it's plenty free space to move it left. 
[image]

 But - again this "but"...
under DRO I mean all this area - let's call it "control panel".
[image]
  It can be moved out of main TAB-view just in place it occupy now, but in the main window area. Program limits better to replace with MPOS, Feed-rate and spindle stae, G-states, remove Home buton and add Stop button instead.
Program settings tab will not fit in this case, but it's can be scrolled down, or better to become separate popup window together with GRBL settings. As I stated before - it's no need to touch this tabs during normal operation.