terjeio/ioSender #252: Feature request: touchprobe pre-check.

calabr, 2022-11-05:
Want to propose feature to check probe connection before probe procedure starts.
  After entering probe page disable to start probing before triggering touchprobe first.
  This option will help to prevent probing if touchprobe not connected or failure.
  Setting should be configurable on the probe page or ioSender setup page, it can be stored as general setup, or as part of probe setup.
  User warning should be displayed before first touch and Start button not active.
  Check request should be re-activated on exit of probing tab.

terjeio, 2022-11-09:
I have added a check on entry of the probing tabs in the [latest edge versions](http://www.io-engineering.com/downloads) - enable in app settings. I might change this to do the check later in the process but that might be a tad more complicated...
The best option would be to implement the probe connected signal in the controller - but that would require hardware changes.

calabr, 2022-11-12:
I'm trying to test it with 2.0.41p7, but didn't find this feature. Instead I found partially implemented probe measure function from [https://github.com/terjeio/ioSender/issues/251](url)

terjeio, 2022-11-13:
Did you enable in _Settinghs: App_?

[image]

calabr, 2022-11-13:
Sorry, I didn't find this checkbox with first look at the options.
  Now I tried it and, unfortunately, have to tell, the current implementation is more annoying, than helps.
  probe check should be done just before probing start - after initial positioning and probe setup. That is why I'm asking to disable Start button. Also it's blocking to use camera for positioning.
 If I'm closing pop-up window it's allow me to enter probing tab without check - I think it's good idea to allow user bypass pre-check if needed, but it should be separate button with clearly recognizable function. If user decide to bypass - he should understand respocability.

calabr, 2022-11-14:
Thanks' a lot! Now it's really works!
The only found - If alert window closed with "Esc" or cross - alert window not closed correctly - window picture is not cleared, but all works correctly.
 the only suggestion "Probe Ready" indicator and probable it's allow to assert probe before rising the alert window.