terjeio/ioSender #253: Feature request: Theming

Xchrisfox, 2022-11-12:
I was hoping we could add easily adjusted themes.

Would love some additional color in my build.

Not a rush but definitely a suggestion.

terjeio, 2022-11-12:
Any suggestions for what should be coloured and with which colour?

Jay-Tech, 2023-06-13:
Hello terjero, 
I created some basic themes for Io Sender there are 4 dynamic  predefined  themes located here:
https://github.com/Jay-Tech/ioSender/tree/themes 

I think the themes are pretty light nothing to crazy and kept the controls and layout the same just a few padding and margin adjustments here and there.  
I have only applied the themes to Iosender XL. at the moment.   The themes  can be asscessed  through  menu/help/themes .  Just starting to test everything now on a actully machine.  Let me know if this is something you are interested in and would like something changed.

terjeio, 2023-06-13:
Great, I'll take a look at this a little later as I will be quite busy for the next few days.

> The themes can be accessed through menu/help/themes.

It would be nicer if made selectable in the _App: Settings_ tab [Main group](https://github.com/terjeio/ioSender/blob/master/CNC%20Controls/CNC%20Controls/BasicConfigControl.xaml) and restored from the [config file](https://github.com/terjeio/ioSender/blob/master/CNC%20Controls/CNC%20Controls/AppConfig.cs) on startup?

Jay-Tech, 2023-06-13:
So it saves the user selection  to the current config that's in place app config if that's the right name.  Restores that selection on launch and yes the theme  selection can be moved to anywhere and easy to do.  The themes are dynamic and swap very quickly. Still like/need  to test on old hardware to make sure they is no ui lag on changing themes  No rush got a few minor details left to clean up like file icon swapping for dark to white to based on ui color thanks

terjeio, 2023-06-22:
I've been experimenting with this a bit now. I tried first to load your version directly but that failed (VS2015) so I decided to import bits of your code in my local master and in the process move the theme handling to the _CNC Controls_ library (from the main applications). Moving it means that it will be shared between apps (currently I have six on top of the libraries) with little code duplication, making maintenance easier. By doing this I ran into a host of different issues, some likely due to my limited knowledge of C#/WPF and some due to limitations in the framework in the version I am using (.Net 4.5.2).

So, how to proceed? Updating the .Net framework to a newer version is an option that may make life easier, but not to a version that does not work with Win7. This since I have several somewhat dated Win7 PCs I use for my CNCs, and I guess other users has as well. Another option is to use a third-party framework (or bits of it), I have found two that may fit the bill, [MahApps.Metro](https://github.com/MahApps/MahApps.Metro) seems pretty advanced but is perhaps an overkill? Interestingly this uses bits of a [third-party library](https://github.com/ControlzEx/ControlzEx/blob/develop/Wiki/ThemeManager.md) for theme handling which may better fit the sender and not require a lot of code changes.
What do you think?

Here are some examples of what I have been experimenting with:

[image]

Theme selection in Settings: App and file icon swapping. Theme selection requires a restart of the app, I am not sure why (trigger is firing for the property but is not catched by the xaml).

[image]

Tabs are upside down, orientation has to be fixed.

[image]

I've removed the drop shadow as felt it just adds "noise". Highligthing looks odd to me and should be improved?

BTW which version of Visual Studio are you using? I am still using VS2015 as I have a licensed version installed.

Jay-Tech, 2023-06-22:
Ok, just went back and  checked it and  maybe I added to many branches in the repo but the working branch is  named themes and that branch  should have all the projects set to 4.52 I recalled you saying you used vs2015 and thought I kept the projects .targeting .net  4.5.2  for this reason  even though  vs2019  tried to update them  to 4.72.  I had issues with library references so I pulled down  nugget packages that matches the 3rd party dlls and this might be causing some issues for you and did not think about it  There is  probably a nuget package manager difference and you would not be able to use the nuggets the way I did .  Yes I agree with the styles should/ would be nice to be shared and  I originally had it at the cnc control level but there are  ui controls  that where fighting me that are shared between xl and normal but loaded a different way so one style would cause the other tool to be different looking between the two versions of apps.  one way around that would be maybe duplicate the control just for xl or normal io and call it something like JogXL  and I believe it was really just the jog control that beat me and  caused me to move  the sttyles to seperate locations .  Using third party stylers or theme makes life a lot easier most of the time.  They range from  mild to really aggresive  Think I had close to 1000 lines of style code it became alot more then I thought it would be when starting this, so a 3rd party sytler might be the way to go moving forward.   ..  I think your reason for staying at 4.5.2 is very valid for this appliation  seeing that is an open source envioment and it being a cnc sender software there might be alot of people still holding on to old PC with win 7 just to use for cnc stuff.
Below are the themes I made and how you select them.  If you used this branch GitHub - Jay-Tech/ioSender at themes (not master its broke atm)  and had the issues let me know and I will remove the nugets and refernce the binaries/3rd party dlls to see if that gets you going.  For work I try to use mainly Vs2019 pro but sometimes have to use VS 2022pro which I really dislike.  Home I use 2019 communty 
  

Went back and made a standard not 100 pecent but close ?            
                                                                                                                                                                                                            Black                                                                                                                                                                                               Dark
   

Light                                                                                                                                                                                         
                                                                                                                                                                                                                                                             White (probably need to remove shadowing hard on with my eyes)                                                                                                    Selection method ( I like your way  better)

                                                                                         

Very aggressive 3rd party styler but very easy to use