terjeio/ioSender #259: Build error

jielemon, 2022-12-12:
A newbie in visual studio.

Downloaded the latest revision of the ioSender source and opened ioSender.sln with Visual Studio 2022. There are 65 errors when building.  Here is the screenshot of some of errors.
[image].

Could you help with that?

Thanks a lot.

Jzhenli, 2023-02-06:
@terjeio I'm newer to this project, and build it error too.  How to install the missing the dependency in the VS IDE?

terjeio, 2023-02-06:
@Jzhenli See [the wiki](https://github.com/terjeio/ioSender/wiki/Building).

Jzhenli, 2023-02-06:
@terjeio Hi, where to find these libs, I check the latest code folder and cannot find them.
[image]

terjeio, 2023-02-06:
You have to download them from the provided links or extract them from the latest [release](https://github.com/terjeio/ioSender/releases).

Jzhenli, 2023-02-06:
Do you mean to download the pre-build version from the release？ I know that will works well, but now I want to know how build them from the VS IDE by myself, and I have met the same error as the topic image shows.

Jzhenli, 2023-02-06:
@terjeio Do you mean to download the pre-build version from the release？ I know that will works well, but now I want to know how build them from the VS IDE by myself, and I have met the same error as the topic image shows.

terjeio, 2023-02-06:
> Do you mean to download the pre-build version from the release?

Yes, that is an alternative to get the libraries if not downloading via the links - copy the relevant files as described in the wiki, they can be found inside the release folder and, IMO, have pretty obvious names.

terjeio, 2023-02-06:
In the root folder of the sender projects I have:

_CNC Controls Camera\bin\Release_ folder with
```
AForge.Video.DirectShow.xml
AForge.Video.dll
AForge.Video.xml
AForge.xml
AForge.dll
AForge.Video.DirectShow.dll
```
Referenced by _CNC Controls Camera_ project. This is a bit odd - cannot remember why now. It may be better to put these in a _aforge_ folder in the root folder of the project and reference the .dll files anew. Or disable the camera as described in the Wiki.

_helix-toolkit_ folder with
```
HelixToolkit.Wpf.dll
HelixToolkit.Wpf.XML
```
Referenced in the _CNC GCodeViewer_ and _CNC Controls Probing_ projects.

_Vector-master\RP.Math.Vector3\bin\Release_ folder with
`RP.Math.dll`
_Vector-master\RP.Math.Vector3\bin\Release\RP.Math_ with
`Vector3.dll`
Referenced in the _CNC Controls Probing_ project.

_websocket-sharp\websocket-sharp\bin\Release_ folder with:
`websocket-sharp.dll`
Referenced in the _CNC Core_ project.

Hope this helps.

Jzhenli, 2023-02-07:
@terjeio Have tried as your method, still cannot build the code. Would you told how to add the Vector lib? Seems cannot find it in the NuGet. The other libs I have installed as bellow image shows.
[image]

terjeio, 2023-02-07:
@Jzhenli Deleting and readding the references to the vector libs failed? Thats odd. Download the source and compile as a separate project and then delete and readd the references to the release build .dlls.

Jzhenli, 2023-02-07:
@terjeio the setps I do as follows,
1. download source code ioSender-2.0.41.zip
2. unzip code and open ioSender.sln with vs2019 (os win11)
3. install below libs with NuGet, the version as bellow image shows
[image]

4. build code and 2 Erros messages
[image]

I don't know the reason why it fails, would help to check the steps? I guess the lib version may have some issues.

Thanks.

terjeio, 2023-02-07:
It seems I have left references to the XCeed toolkit in the published source - a missing commit on my side.
Replace the two files in the attached zip in the _\CNC GCodeViewer\CNC GCodeViewer_ folder and try again.

[CNC GCodeViewer.zip](https://github.com/terjeio/ioSender/files/10673136/CNC.GCodeViewer.zip)

terjeio, 2023-02-07:
_ToolIndicator.cs_ is experimental code - exclude it from the project by right clicking on the file in the Solution Explorer and select _Exclude From Project_.

Jzhenli, 2023-02-08:
@terjeio thanks for your reply, now it's OK.