terjeio/ioSender #260: Use G59.3 As TLR - tool length reference

LoganFraser, 2022-12-12:
Basically add an option to use the G59.3 directly as the TLR - Tool length reference. That way it's persistent between resets, and tools are measured compared to a static machine space point.

I don't know if this is a grblhal core thing or if it could just be set within iosender as T98 or something (since T99 is often used for 3d probe)

terjeio, 2022-12-12:
It is a [core thing](https://github.com/grblHAL/core/blob/master/tool_change.c), the sender handles all tool change modes (except ATC toolchanges which are 100% transparent to the sender) the same way.

Associating a tool with the probe is new info for me, any links or further information on how this is supposed to work?

LoganFraser, 2022-12-12:
Basically, the PrintNC group runs G6T99 in to "pick" the probe, and touch off and set the tool offset reference.  But if a reset occurs, the initial TLO is lost and set to the next tool in the series instead which means G54 Z changes.  So the hope is to have a set TLR that survives a reset.

terjeio, 2022-12-12:
>  So the hope is to have a set TLR that survives a reset.

Ok, can you try with moving [these lines](https://github.com/grblHAL/core/blob/d10804433c2b1060bb8ad6c5fa04218c2cf435a8/system.h#L232-L233) to down below line 359 (after alarm)?
I am not able to test this myself as I am away.

LoganFraser, 2022-12-13:
I'll see if I can get a teensy environment setup for compiling tonight and give it a test.  Thanks.

Dietz0r, 2022-12-15:
Would it be possible to set a manual $TLR? 
When we could set it at a known location that would basically provide a workaround route for just setting $TLR after homing.
We've tried issuing a $TLR (since the documentation implies this is possible) over the MDI and it responds with OK but doesnt give the tool referenced flag.