terjeio/ioSender #261: Heightmap with modal gcodes

tetref, 2022-12-22:
I don't know if I'm writing this correctly:
I have a post processor that creates a circle with modal values X Y I J such that it logically omits some, eg G3X0I10J0. If I apply Height map to such code, error 33 occurs on the first line with G3.
It seems to replace or add the missing value X to Z. So the line looks something like this:
G3X17Z0.196I-10J0
I tried different gcodes. I have no problem with this code before applying. Only after setting Y as nonmodal is everything fine. The problem is with all versions of the program.

terjeio, 2022-12-23:
Do you have an example (including a height map file)?

tetref, 2022-12-23:
I will add the heigtmap and the modified file later:

Modal codes:

%
G21
G90
M6
S10000M3
G0Z6.000
G0X10.000Y5.000
G1Z-0.100F300.0
G17
**G3X0.000I-5.000J0.000F100.0
G3X10.000I5.000J0.000**
G0Z6.000
G0X10.000Y5.000
G0 X0.0000Y0.0000
M30

Nonmodal codes:

%
G21
G90
M6
S10000M3
G0Z6.000
G0X10.000Y5.000
G1Z-0.100F300.0
G17
**G3X0.000Y5.000I-5.000J0.000F100.0
G3X10.000Y5.000I5.000J0.000**
G0Z6.000
G0X10.000Y5.000
G0 X0.0000Y0.0000
M30

tetref, 2022-12-27:
Original gcode:

%
G21
G90
M3 S10000
G0 Z6.000
G0 X20.000 Y10.000
G1 Z-0.100 F200.0
G17
G3X0.000I-10.000J0.000
G3X20.000I10.000J0.000
G0 Z6.000
G0 X20.000 Y10.000
G0 X0.0000 Y0.0000
M30

Heightmap:

<?xml version="1.0" encoding="utf-8"?>
<heightmap MinX="0" MinY="0" MaxX="20" MaxY="20" SizeX="5" SizeY="5">
  <point X="0" Y="0">0</point>
  <point X="0" Y="1">0.029</point>
  <point X="0" Y="2">0.032</point>
  <point X="0" Y="3">0.06</point>
  <point X="0" Y="4">0.064</point>
  <point X="1" Y="0">0.049</point>
  <point X="1" Y="1">0.072</point>
  <point X="1" Y="2">0.073</point>
  <point X="1" Y="3">0.092</point>
  <point X="1" Y="4">0.105</point>
  <point X="2" Y="0">0.081</point>
  <point X="2" Y="1">0.109</point>
  <point X="2" Y="2">0.133</point>
  <point X="2" Y="3">0.132</point>
  <point X="2" Y="4">0.143</point>
  <point X="3" Y="0">0.147</point>
  <point X="3" Y="1">0.162</point>
  <point X="3" Y="2">0.169</point>
  <point X="3" Y="3">0.174</point>
  <point X="3" Y="4">0.179</point>
  <point X="4" Y="0">0.181</point>
  <point X="4" Y="1">0.193</point>
  <point X="4" Y="2">0.2</point>
  <point X="4" Y="3">0.198</point>
  <point X="4" Y="4">0.196</point>
</heightmap>

Gcode after Apply:

G21
G90
S10000M3
G0Z6
G0X20Y10Z6.2
F200
G1Z3.15
G1Z0.1
G17
G3X19.01Z0.093I-10J0
G3X16.235Z0.082I-9.01J-4.339
G3X12.225Z0.059I-6.235J-7.818
G3X7.775Z0.025I-2.225J-9.749
G3X3.765Z-0.01I2.225J-9.749
G3X0.99Z-0.037I6.235J-7.818
G3X0Z-0.068I9.01J-4.339
G3X0.99Z-0.062I10J0
G3X3.765Z-0.052I9.01J4.339
G3X7.775Z-0.032I6.235J7.818
G3X12.225Z0.011I2.225J9.749
G3X16.235Z0.062I-2.225J9.749
G3X19.01Z0.088I-6.235J7.818
G3X20Z0.1I-9.01J4.339
G0Z6.2
G0X20Y10Z6.2
G0X0Y0Z6
M30

[image]
[image]
[image]

terjeio, 2023-01-07:
I belive I have fixed this, but I want to verify with a more comprehensive piece of gcode before release. This has to wait until I am back home later in January. Until then you may use _File > Transform > Arcs_ to lines before applying the map as a workaround.

terjeio, 2023-01-27:
Can you try the latest edge version?, available from [here](http://www.io-engineering.com/downloads).

tetref, 2023-01-27:
Neither version 2.0.42p0 can be started, it reports an error.
[image]

terjeio, 2023-01-27:
I've tested the standard version after downloading the zip and it works for me.
Can you translate the error message?
Which controller are you using?

tetref, 2023-01-27:
Info from 2.0.41

[VER:1.1f.20221101:]
[OPT:VNMZSL,35,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,ES,TC,SED]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:STM32F411]
[DRIVER VERSION:221014]
[BOARD:BlackPill]

The index is outside the field bounds. Probably the correct translation is : "Index Out of Range Exception".

terjeio, 2023-01-27:
Thanks, that helps. Fixed versions uploaded now.

tetref, 2023-01-27:
Version 2.042p1 seems to work ! After loading the gcode, the circle appears as a hex, and after applying the heightmap, it appears as a "perfect" circle. Perfect, seems to be fixed, thanks...

closed 2023-04-29 by terjeio