terjeio/ioSender #262: TLO probing inacuate offset

calabr, 2022-12-22:
sender 2.0.41p8
 In TLO probing circle offset aplyed with **G10L20P1Z0** which use current position instead of probbing result. I's give some speed and acceleration depended error relaive to actual probbing results. 
In my case I get 0.02 - 0.04 mm difference to edge-finder Z mesurement results with reasonable slow feed rate.
 I also found, G10L20P1Z0 applyed twice ( shown in terminal window)
@terjeio could you please set the same algoithm as for egde finder

terjeio, 2022-12-23:
Is this in from the _Center finder_ tab or the _Tool length offset_ tab?

FYI I am on holiday and do not have access to a machine to test with so I cannot look into this before I am back home.

calabr, 2022-12-28:
Hello @terjeio  Wish you happy New Year!
  It's from TLO tab. Edge finder and center finder works correctly.

terjeio, 2023-01-07:
Happy new year!

You are correct in that the current position is incorrectly used instead of the probed position. I'll fix this and retest when I am back home.

> I also found, G10L20P1Z0 applyed twice ( shown in terminal window)

I cannot immediately see this happening from looking at the code, so this has also to wait until I am back.

terjeio, 2023-01-27:
I have now uploaded a new [edge version](http://www.io-engineering.com/downloads) with a fix.

calabr, 2023-02-15:
Verifyed with 2.0.43p3

closed 2023-02-15 by calabr