terjeio/ioSender #268: REQ: Option to enable timestamp pre-pending in console feedback view

5ocworkshop, 2023-01-19:
Thanks again for all the great work on grblHAL and IOSender.

I was working with some folks on the PrintNC Discord today investigating some odd behaviour around spindle on delay and we were comparing the actual ramp times to various target RPMs with the trigger settings for the alarm.  It occured to me while doing this that in some instances it could be really helpful to have date/time stamps prepended to the log entries in a fixed width format so that we could see the exact seconds/milliseconds the controller is reporting certain things.  It would help during debug/QA work.

Is this something that can easily be added at the UI level with a tick box to enable or disable?  I'm afraid I can't offer to assist in coding it but I am happy to test it and report back.

terjeio, 2023-01-20:
Can this grblHAL [plugin template](https://github.com/grblHAL/Templates/tree/master/my_plugin/Real%20time%20report%20timestamp) be used? It adds a timestamp to the real time report resettable by M101.

`<Idle|MPos:0.000,0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZA|TS:9:04,0869>`

5ocworkshop, 2023-01-23:
That looks very helpful for debug.  Thank you!

terjeio, 2023-01-25:
Some drivers has real time clock (RTC) available already - mostly networked drivers. This is set from ioSender on connect.
`$RTC` reports the current value when the clock it is set, e.g `[RTC:2023-01-25T20:11:09]`

Do you need millisecond resolution for the timestamp and can/should the date part be skipped?

5ocworkshop, 2023-01-29:
I don't think millisecond is necessary for most testing and the date can probably be skipped without any significant loss.  I'm not sure if the Hal2000 or the FlexiHAL have RTC, will check.

terjeio, 2023-01-29:
At least the Teensy has a RTC, running off a 32KHz crystal. IIRC it is possible to run a RTC off the main oscillator with some loss of accuracy - and FYI the STM32F4xx driver has RTC support, but may need a [code change](https://github.com/grblHAL/STM32F4xx/blob/dda9faed5a0f298595ac6cf219d3f13a2b543d09/Src/driver.c#L1982-L1990) to select the oscillator source if a 32KHz source is not available.

I'll add a M-code parameter to the plugin for switching timestamp between elapsed time and RTC.

terjeio, 2023-01-30:
[RTC support added](https://github.com/grblHAL/Templates/blob/7b5d083ed37b50bd127756a415580f62da261336/my_plugin/Real%20time%20report%20timestamp/my_plugin.c#L8-L14). ioSender sets the RTC when connecting to the controller, for others set it with the `$RTC` command.

andrewmarles, 2023-01-30:
FlexiHAL will need a different clock config to enable RTC as the external XO is not present (pins are used for other things).  I will work on adding support and a PR.