terjeio/ioSender #270: Lathe mode DRO display issue

docwelch, 2023-02-03:
Using the latest release, 2.0.41, when I enable the lathe mode and restart the app, the DRO axes that display are X and Y instead of X and Z. 

Also, the DRO always shows to be in radius mode even when changed to diameter mode (restarting does not fix this).  I do not have my lathe set up yet so do not know if this is just an error in the display or if it is an error in how the program behaves.

terjeio, 2023-02-03:
Wrong axes showing is due to a regression - easy to fix.
Radius/diameter mode is normally set from the controller, if is a grblHAL controller set the _Mode of operation_ to _Lathe mode_ ($22=2). The Diameter/Radius mode is selected with G7 and G8 respectively, try that from the MDI. This works for me.
If a legacy Grbl controller - then I have to go back and check the code...

I'll upload a new Edge version when you have checked Diameter/Radius mode.

docwelch, 2023-02-03:
I am using an ESP32 based controller so setting the mode in the controller and using G7/G8 does work as it should. 
I'll watch for the new Edge version

Thanks for your great work and quick response!

terjeio, 2023-02-04:
New [Edge version](http://www.io-engineering.com/downloads) with fix uploaded.

docwelch, 2023-02-04:
Thank you, working as it should.

closed 2023-02-04 by docwelch