terjeio/ioSender #272: Project Document

Jzhenli, 2023-02-08:
Hi, I'm beginner, seems there's no doc for this project, would you give a brief introuduce for each part?

[image]

If you have any other doc to help understand this project, pls also let me know.
Thanks.

terjeio, 2023-02-08:
I do not have any documentation for the code other than the code itself.

The app is layered and the most important/basic thing about it is that it is based on the [MVVM coding pattern](https://en.wikipedia.org/wiki/Model%E2%80%93view%E2%80%93viewmodel).

The _CNC Core_ project is the foundation, it handles communication with the controller, parses data from it and sends data to it via the different streams available. The data parser implements the basic viewmodel that delivers information to the UI. There is also a gcode parser that tokenizes the gcode for use by the 3D viewer and gcode converters. There are no UI components in this project.

The _CNC Controls_ project contains all the "standard" UI controls and has higher level classes/methods for connecting to the controller and handling settings and settings UI.

The _ioSender_ project is the top layer that creates the UI and brings it all together. There are three versions of this top layer - I have one customized version I use for my CO2 laser which uses some additional UI controls not part of the published code.

The CNC AppLaunch is a small application that allows external apps (such as [Vectric VCarve](https://github.com/terjeio/ioSender/wiki/Vectric-Direct-Output)) to start the app and send a command to it for opening a gcode file.

There rest of the projects contains functionality/UI components that are optional for building an app, but surely useful.

The _CNC GCodeViever_ contains UI components for displaying gcode and presenting configuration options for itself. It uses the tokenized gcode for rendering.

The _CNC Controls Probing_ contains UI components and functionality for probing.

The _CNC Controls Lathe_ contains UI components and functionality for interacting with a lathe.

The _CNC Controls Camera_  contains UI components and functionality for interacting with a camera, used for positioning and measurements.

The _CNC Converters_ contains functionality for conversions from other formats to gcode.

The _CNC GCodeViever_ contains functionality and an UI control for transforming gcode to allow it to be used with a dragknife. It uses the tokenized gcode.

The filenames and other names (classes, methods, arguments, variables etc.) are mostly self explanatory - at least for me...

Jzhenli, 2023-02-08:
ok, thanks for your reply, it's really helpful.

closed 2023-02-13 by Jzhenli