terjeio/ioSender #273: Can't zero the DRO X value after probing w/o ioSender Unhandled exception occurred

R-Ronald, 2023-02-12:
If I probe the bottom edge of a piece of stock with Probe Z checked, it works fine. The operation sets both the Z height and the Y position where it touched the part to zero in the DRO and G94.

Since this is a long piece of stock, I then jog along the x-axis to visually pick the point where I want X to be zero. Hit the Zero button beside X and the code generates an "ioSender Unhandled exception occurred" and exits.

Not the end of the world, since the program can be restarted. The TLO and and Home machine position are not lost. After a restart, it is willing to accept the button press to zero the X axis DRO value (and the G94 workspace).

terjeio, 2023-02-13:
Which ioSender version?
Is this issue repeatable?

R-Ronald, 2023-02-14:
ioSender XL is at version 2.0.41

For me, this error happens every time (is perfectly repeatable). I've also seen it after other probing operations, but the only one  that I know is repeatable is using the edge finder to probe a single edge (always tested on the bottom side of the X axis) along with Z.

terjeio, 2023-02-14:
I am not able to replicate with the [latest edge version](http://www.io-engineering.com/downloads). Can you with try that?
If this fails to then:
If you do not jog the X-axis before zeroing X does it still fails?
Zeroing any other axis is ok?

BTW I am using Win10.

R-Ronald, 2023-02-14:
There was no change from using the Edge version.
It does not matter if one jogs or not, prior to pushing the "zero" button.
I didn't answer your last question about the other axes...because something else jumped out. I think this will help point in the exact direction of the problem.

What I noticed was: On the grbl tab, everything works properly. On the probing tab, it does not. In other words, the zero button next to the "X" DRO box works fine if it is on the grbl tab...but not for the one on the Probe tab. 

I'd bet that your attempt to replicate used the grbl tab. So, the problem did not occur.

I also noticed that the error message is not exactly what I documented. The full message is:
Dispatcher Exception. Unhandled exception occurred.
Object reference not set to an instance of an object.

terjeio, 2023-02-14:
I overlooked the DRO in the probing tab - it was not correctly initialized. New edge versions just uploaded with fix,
Note that I have disabled position entry in this DRO, it behaves strangely and I need more time to sort that out.
Oh, and the issue had nothing to do with the probing sequence at all...

R-Ronald, 2023-02-14:
Thanks.

After my last response, I realized that I only used that page/DRO after doing a probe and I'd never tested setting zero without dong the probe. 

Thanks for the quick fix!

closed 2023-04-29 by terjeio