terjeio/ioSender #274: Enhancement Request: Don't start operation for a tool if it will hit soft limits

R-Ronald, 2023-02-12:
I've run several jobs now where I've hit the soft limit for Z during gcode execution. The tool is moving down (typically plunging or drilling), but soft limits for X stop it before it reaches the programmed height. So, the operation stops. Nothing wrong with this behavior.

However, it would be preferable if ioSender checked the operation was possible prior to starting it. Fusion 360 puts ZMIN values in its gcode header, so the gcode would not even need to be line-by-line parsed for a comparison against the results from the measured tool length (adjusted by the TLO and workspace zero). 

IMHO, much better to find out you can't complete the cut before spinning up the spindle and starting it. 

I removed the skateboard bearing on my PrintNC today and gained about 10mm more negative Z travel, so I plan to never again need this feature for myself. But, I think it would be useful.

terjeio, 2023-02-13:
> However, it would be preferable if ioSender checked the operation was possible prior to starting it.

I have no plans to add sanity checks on running gcode in the sender, IMO too much work for little gain.

calabr, 2023-02-13:
@R-Ronald  You can run your program in check mode and controller itself will do all sanity according to current setup and position.
 What would be helpful from IoSender side - clear readable execution results report with execution results, error codes, last executed program line and content, execution time, any intervention by user, like pause, feed override events with timestamp and effective program string etc. 
 Unix-stile log file with supporting window in GUI looks good for me.

terjeio, 2023-02-13:
> You can run your program in check mode and controller itself will do all sanity according to current setup and position.

This is not good enough when there are tool changes involved that is likely to change offsets that check mode cannot know anything about.

R-Ronald, 2023-02-14:
i.e., before you actually go far enough to generate the offset for the tool in a specific operation, check mode is missing data about how the Z axis will need to move.