terjeio/ioSender #276: 3D view not rendering properly

brex480, 2023-02-16:
First off I'm sorry if I am missing an obvious step. I saw a section in the wiki which might be the answer and made sure what to do with it. Anywho, my issue is the 3d view panel is unusable. Rotating the image around I can see glimpse of the file but it is like the contrast is turned up all the way on an old tv. I have tried several different releases running on windows 11. Thank you

[image]

phil-barrett, 2023-02-16:
Try turning off the grid.
[image]

brex480, 2023-02-16:
Should have mentioned I have tried toggling off and on all the buttons there and in the settings panel. I can see changes happening but always still blurred like looking through a extremely fine mesh.

terjeio, 2023-02-16:
Have you checked _Black background_ in _Settings: App_? If so I have to find or write a new color picker for that to become useful.

brex480, 2023-02-16:
In the first picture yes it was selected but that it not the cause of the underlying issue. He it is unselected.
[image]

terjeio, 2023-02-16:
Can you attach the file (zipped) to a comment?

brex480, 2023-02-16:
Guessing you are asking for the gcode file. Its just something I popped out of fusion 360 real quick using openbuilds post to test out iosender. It opens fine in openbuilds control and gsender.  Also tried a file that came from openbuilds with the same results. 
[1001.zip](https://github.com/terjeio/ioSender/files/10760699/1001.zip)

phil-barrett, 2023-02-16:
It works fine for me on Win 10 using 2.0.40 and 2.0.41 using default ioSender settings.
[image]

terjeio, 2023-02-16:
Renders ok for me, it seems there are many short movements so check _Min distance_ and possibly also _Arc resolution_:

[image]

brex480, 2023-02-16:
> It works fine for me on Win 10 using 2.0.40 and 2.0.41 using default ioSender settings. Wonder if it's a graphics card issue? [image]

Winner winner, forced it to use the actual GPU and now works perfectly. Thanks for the help. I am loving iosender even when I couldn't get the viewer working.

phil-barrett, 2023-02-16:
What exactly did you do?  (this is for others who run into similar situations)

brex480, 2023-02-16:
On windows 11 you will go to display settings and click graphics. From there you will have to add iosender.exe to a list of applications. Then you can click the options menu next to iosender and select high performance. I have had to do it for other programs in the past but didn't cross my mind today.

closed 2023-02-16 by brex480