terjeio/ioSender #277: Feature request: Portrait Support.

LoganFraser, 2023-02-24:
There's a bunch of low-cost portable touchscreens with 1080p (1920x1080) that would be great mounted to a CNC in portrait mode.  Would it be possible to have all the 3 panels loaded at the same time vertically for that display orientation - 3d view, gcode and console?

terjeio, 2023-02-25:
You can try that?

The UI layout for the center part of the _Grbl_ tab [is here](https://github.com/terjeio/ioSender/blob/6a7f84b394e0469ccd15ea207d5024cd251f9df5/ioSender%20XL/ioSender%20XL/JobView.xaml#L68-L81).  Unwrap and remove the tab control and replace the dock panel with a vertical stack panel should work. You may keep the tab control without the console if you want both the 3D view and the program view. The jogger [is added here](https://github.com/terjeio/ioSender/blob/6a7f84b394e0469ccd15ea207d5024cd251f9df5/ioSender%20XL/ioSender%20XL/JobView.xaml#L56), and can be moved to the new stack panel.

LoganFraser, 2023-02-25:
I appreciate your pointing out the locations. I'll see if I can get someone from the PrintNC discord to follow up.  I'm an electrician/hardware design guy, my programming skills are extremely limited.  Thanks for the reply.  :)

mintracer, 2023-03-20:
Hi Terje,

so this is my first attempt in changing an UI in Visual Studio... My first touch with this program :)
I changed the JobView.xaml to a Grid Layout - 2 Colums and 3 Rows 
I hope you like it and can give me some hints how to make it better :) 

btw. The Gcode is just a double bottle opener... :D

[image]
[image]

terjeio, 2023-03-20:
Looks good - but it is wide enough for the probing tabs?

Instead of two columns at the top three will be better? At least with more than 3 axes configured? IMO _Program limits_ could be moved to the center column.

mintracer, 2023-03-20:
todays progress... :)

[image]

terjeio, 2023-03-21:
I have to admit that I am old fashioned and do not fancy such colorful displays. Users younger than me migh have a different opinion - and I am ok with that. Hopefully someone else can comment too.

mmaczollek, 2023-03-25:
I would love a dark theme to the program. I spend a lot of time on computers for work and the light background gives me migraines. @mintracer, can you share that UI somehow? It looks really great.

mintracer, 2023-03-25:
> I would love a dark theme to the program. I spend a lot of time on computers for work and the light background gives me migraines. @mintracer, can you share that UI somehow? It looks really great.

it´s a really early preview... since i´m not a professional coder as Terje is... I´m learning all the wpf theming... so it will take a while before any "sharing" will be possible.. and offcourse Terje has the last word on it... it is his "baby" ;) 

@terjeio is there a central place for all the coloring in wpf? such as css on websites?

terjeio, 2023-03-28:
> is there a central place for all the coloring in wpf? such as css on websites?

I have not read up on [theming in WPF apps](https://duckduckgo.com/?t=ffab&q=wpf+theming&ia=web) so I do not know what is required nor how it is done

axg20202, 2023-10-04:
Hi all. Sorry to revive this old thread, but is anyone able to share a portrait layout (source files, preferably), in default colours as per first image above?

LoganFraser, 2023-10-05:
There's a touch focused fork of iosender that has more direct support for portrait.  My machine is currently in pieces so I haven't looked at it in detail yet but I'm impressed with what I have seen.  https://github.com/Jay-Tech/ioSenderTouch/tree/IOSenderTouch

LoganFraser, 2023-10-05:
Didn't mean to close it.  ;)

axg20202, 2023-10-06:
Thanks. The issue is that I'm not using a touchscreen. Willing to consider switching to one though if it really is a good move.

phil-barrett, 2023-10-06:
I'm looking at using a touch screen.  A lot of people hate the idea of hooking up a PC to their CNC machine but love the idea of the Masso controller which is effectively the same with a touch screen.  I think that using a breakoutboard, microPC and touchscreen could be about 75% of the cost of a Masso and way more flexible.  Will be protoyping this fairly soon.

axg20202, 2023-10-06:
Are there touchscreens that are recommended or those to avoid? I've seen some comments on avoiding capacitive screens, but I think those are the affordable ones.

phil-barrett, 2023-10-06:
actually, it is resistive touch that is slow an imprecise.  cap is better.  

I am looking for a windows touch screen that works with a "stock" miniPC.  I believe they work by using a USB connection to emulate mouse/keyboard.

axg20202, 2023-10-06:
OK, good to know. I hope a capacitive screen won't go nuts as soon as my vfd fires up. I've seen quite a few all in one pc that have the guts of a mini pc built into the touchscreen too. I.e. Hp G2 touchscreen, Dell made some too.

phil-barrett, 2023-10-06:
Yes, the all-in-one PCs will probably work ok but they tend to be expensive. I am seeing miniPCs that are in the $200-300 USD range that should work pretty well.

LoganFraser, 2023-10-06:
I'm just using one of the portable USB-C powered 15" touchscreens available on Amazon.  It works well enough.

axg20202, 2023-10-07:
I've decided against touch screen. Can anyone help me out with editing the job view xaml of the standard size iosender? I really like the first portrait layout shown towards the top of this thread. Even that job view file would be a huge help for me to get started on tweaking it further.

axg20202, 2024-01-03:
Just bumping this thread. Terje, is this something that you could support please? Being able to view ioSender in portrait would be hugely helpful but I am completely at sea when it comes to editing the code to achieve this.

andrewmarles, 2024-01-03:
Hopefully @terjeio it is ok to post, but a user has done a beautiful port of the iosender codebase that is optimized for touch and has portrait support.

https://github.com/Jay-Tech/ioSenderTouch