terjeio/ioSender #279: Heightmap does not start probing

S2000Stefan, 2023-03-01:
Hello terjeio,
wanted to use the surface scan once this morning. Unfortunately, I have to say that nothing happens after I press start and the machine automatically moves to the bottom right corner. The probing works perfectly otherwise.
The controller is a F303 Blackpill.
Also, nothing is displayed in the fly out menu for the machine coordinates. In the 3D window the display is available but not in the fly out menu.
Greetings Stefan

S2000Stefan, 2023-03-01:
I take back all panic. :)
I was able to successfully complete my milling programme with heightmap this afternoon with version 2.0.38 Final.
After that I had time to test the new Edge version 2.0.42p4 and can also say that the heightmap scanning works there.
Just to inform you, the version that did not work was 2.0.41p3 or an operating error. :)
So all is well.
The only thing I noticed was that although I placed X0 Y0 in the middle of my workpiece, ioSender took the bottom left as the zero point when measuring Z0. It was a bit confusing because in the middle of my workpiece Z0 suddenly had +1.02mm. But the milling was perfect afterwards.
The error in the fly out menu exists in all versions that I have tested. No display of the machine coordinates.
Greetings Stefan

terjeio, 2023-03-01:
> The only thing I noticed was that although I placed X0 Y0 in the middle of my workpiece,

I'll have to check this. Your gcode was centered around 0,0 (going negative)?

> The error in the fly out menu exists in all versions that I have tested.

Odd, it works for me:

[image]

Are the fields blank? Since it works in the 3D viewer it cannot be due to a missing flag in `$10` setting (_General_, _Status report options_)...

S2000Stefan, 2023-03-01:
> I'll have to check this. Your gcode was centered around 0,0 (going negative)?

I first set the machine zero point in the middle of my workpiece. Marked in red. X0 Y0 Z0
After scanning, Z0 was no longer zero but Z1.02.
That confused me a little. 
If I understand correctly, after the surface scan, ioSender sets Z0 at the bottom left although my zero point is in the middle. Marked in orange. Perhaps this is also a problem of understanding on my part.
Anyway, the milling worked wonderfully. 
I hope it's clearer now. 
[image]

For me it looks like this. :)
[image]

terjeio, 2023-03-03:
New [edge version](http://www.io-engineering.com/downloads) is up.

When x0y0 is inside the job and _Set Z = 0 at X0Y0_ is checked a probe at 0,0 is added at the end and the map is offset accordingly.
I could have calculated the offset by using the map but decided against it. When testing with a sloped PCB the difference was less that 10 micrometers, but is likely to be larger if rougher material is probed.

The machine position issue was odd, it is ok with 3 axes but not 4, and there is no obvious connection between the different fields. Looks like it is some framework issues at play, likely timing related. I managed to find a workaround so should be ok now.

S2000Stefan, 2023-03-03:
Thank you for the, as always, very quick solution to my extraordinary problems. I will probably test the new edge version on the machine this afternoon and report back to you.

Because of the machine position, I can only say that it worked in previous versions. I'll have to check which version the problems started with. I can also take a look today at noon.

S2000Stefan, 2023-03-03:
Terjeio, what can I say, Edge 2.0.42p5 works wonderfully. Thank you for that.

For information about the machine coordinates, I can say that the last working version I had on the milling machine was Edge 2.0.38p3. As of version 2.0.38 final, there is no longer any display in the fly out menu. 
It might help you a little. 

I still have one suggestion about the hightmap. Would it be possible to bind the scanning of the z height to the last starting point? I mean like this: If the starting point is 1mm above the workpiece Z+1.0mm, the next scan is 0.5mm lower on the workpiece, so the next starting position is Z+0.5mm,
the next scan is again 0.5mm lower on the workpiece, so the next start position is Z0.0mm and so on. The starting point always follows the contour of the workpiece plus the preset starting point. The advantage would be a faster scanning of a very contoured workpiece and you could scan workpieces with very large height differences. I hope you can follow my explanation.
It's just a suggestion, you can think about it. :)
Greetings Stefan

closed 2023-03-03 by S2000Stefan