terjeio/ioSender #280: feature request: 3d probe operations order (Z first)

LoganFraser, 2023-03-03:
Currently when selecting "include z", Z axis is probed last, which means z height needs to be very accurately set before probing to ensure the edge is in the correct place on thinner materials.  Is it possible to have a toggle that probes Z first (or make it the default), so then the height of the X and Y probe relative to the material can be more easily adjusted?

terjeio, 2023-03-03:
_Probe Z_ first with setting an offset, adjust _Dept_ to match thickness of material, e.g. 50%, plus Z position displayed below and then probe the corner?

LoganFraser, 2023-03-03:
Yes, that sounds ideal.

terjeio, 2023-03-04:
> Maybe I can turn that into a macro then

IMO, not easily. After the initial Z probe Z distance to move before probing the corners depends on the probe result. At least you have to enable expression support in the controller, but I am not sure that will be enough.