terjeio/ioSender #283: Crash when open some GRBL settings

nickshl, 2023-03-29:
Steps:
1) Go to "Settings: Application"
2) Set "Lathe mode" to "Radius" or "Diameter" then save settings.
3) Close ioSender and open it again.
4) Go to "Settings: GRBL"
5) Open "Limits"
6) Click "5 Invert limit pins XYZ" or "18 Pullup disable limit pins".
Result: Unhandled error occur.
It also happens with some other settings.

terjeio, 2023-03-30:
Try the [latest edge version](http://www.io-engineering.com/downloads).

FYI to enable lathe mode in the controller set $32=2, this unlocks some lathe specific gcodes.

nickshl, 2023-03-30:
Bug present in in this version: https://github.com/terjeio/ioSender/releases/tag/2.0.42
Looks like bug doesn't exist in ioSender XL Edge 2.0.43p1.zip
May be it time to make another release?

By the way - "Parting" is grayed out after startup. After press once  jog control X+ or X- it become available. Is there a reason or is it another bug?

terjeio, 2023-04-29:
> May be it time to make another release?

The edge versions are prereleases - I'll add the changes in the next full release.

> By the way - "Parting" is grayed out after startup. After press once jog control X+ or X- it become available. Is there a reason or is it another bug?

It is another bug - fixed in the latest edge versions.

> Another question: is there any RPM window that show actual RPM from encoder?

No. However, when actual RPM is available it is shown in the current _RPM_ field (instead of the programmed speed).

closed 2023-08-26 by terjeio