terjeio/ioSender #288: Feature Request: disable motors by M command.

RDS356, 2023-04-20:
I believe that almost users make $1=255 or just do not wiring ENABLE/DISABLE on your external driver. In this case this command is very useful, because sometimes when the machine is stopped for long time it is interesting to disable the motors to avoid, among other things, unnecessary heating.

terjeio, 2023-04-21:
[This plugin template](https://github.com/grblHAL/Templates/blob/master/my_plugin/Stepper%20enable%20control/my_plugin.c) can be used? In the sender you can add a macro disable the motor(s).

terjeio, 2023-04-21:
Copy _my_plugin.c_ to the main folder, remove `grbl/` from [this line](https://github.com/grblHAL/ESP32/blob/f3b73ab4cb7f640724e29f606704f02dbce3d25c/main/CMakeLists.txt#L208) and recompile/flash.

Most drivers just need recompile after adding _my_plugin.c_ - will add a note somewhere for the ESP32 driver.

RDS356, 2023-04-21:
Solved, thank you again.

closed 2023-04-21 by RDS356