terjeio/ioSender #289: Discussion: temporary machine limits or configurable 'exclusion' zones?

majdko, 2023-04-21:
I'm not sure if it's interesting to anyone else, but I was wondering about the potential to add the ability to define temporary machine limits, say, per a loadable file and/or text input. 

After bringing this up on the pnc discord, someone mentioned an interest in exclusion zones and that might be an overlapping concept. 

Essentially, it would be neat to redefine machine limits on the fly. For example: If I was undertaking an operation with protruding work holding/parts that I really didn't want to hit, it seems helpful. 

Does anyone have any thoughts on this?

terjeio, 2023-04-29:
A grblHAL plugin implementing a custom M-code is the correct way to do this?
Or do you want the check to be performed on file open only?

terjeio, 2023-04-29:
Added [struct to grblHAL](https://github.com/grblHAL/core/blob/master/changelog.md#20230429) that allows modifying jog and soft limits from plugin code.

MARIOBASZ, 2023-04-30:
Sorry if I am repeating your idea, maybe I did not understand correctly: exclusion zones seem interesting to me, especially the depth limit in Z to protect the support bed of the cnc.