terjeio/ioSender #290: IOSender crash when start with door opened.

RDS356, 2023-04-22:
Using the feature 'OPTION(SafetyDoor "Safety door enable" ON)' if you start iosender and this condition is true, it crash.

[image]

[image]

terjeio, 2023-04-27:
The [latest edge version](http://www.io-engineering.com/downloads) 2.0.43p2 should be ok - you can try it?

RDS356, 2023-04-27:
> The [latest edge version](http://www.io-engineering.com/downloads) 2.0.43p2 should be ok - you can try it?

Hi Terje.
Yes, off course i can.

RDS356, 2023-04-27:
The crash problem has solved, but another strange think is happening.
After the process the IOsender is reporting in inches, but is not his my option.
Look the pictures bellow , the sequence of open, reset and jog.

If I start with door closed IOsender do not report in inches.
Thank you.
[image]
[image]
[image]
[image]

terjeio, 2023-04-28:
Oops, door close behaves differently depending on parking mode on/off. I have uploaded a new edge version that hopefully works...

RDS356, 2023-04-28:
> Oops, door close behaves differently depending on parking mode on/off. I have uploaded a new edge version that hopefully works...

Worked perfectly... Thank you Again.
Will new version of IOsender come with this correction ?
Regards, 
Reinaldo de Souza

terjeio, 2023-04-29:
> Will new version of IOsender come with this correction ?

Yes, edge versions are prereleases.

closed 2023-04-29 by RDS356