terjeio/ioSender #291: The cycle start clicking has no effect

Creat-new, 2023-04-26:
When I open it, clicking on manual feed can run normally, but clicking on cycle start does not work. I have tried restarting, reconnecting, and other methods. Can you tell me why? Thank you very much

phil-barrett, 2023-04-26:
Do you have a GCode program loaded?

Creat-new, 2023-04-26:
yes，just like this
[image]
[image]

Creat-new, 2023-04-26:
I tried to directly control with the G0 Z5 command and found that it could be controlled. I wonder if the iosender did not send any commands @phil-barrett

phil-barrett, 2023-04-26:
maybe some details would help - which processor? any plug-ins?  How is the controller set up? (external buttons, limit switches, etc...).

RDS356, 2023-04-27:
Is there any msg on console ?

terjeio, 2023-04-29:
> I wonder if the iosender did not send any commands

A star `*` is shown in front of each gcode block sent, this changes to `ok` when the controller acknowledges reception.
If you enable _Agressive buffering_ in _Settings App_ the sender will fill the controller buffer before waiting for acknowledge from the controller. A grblHAL controller typically has a 1Kbyte buffer, legacy Grbl only 128 bytes.
Note that Grbl v0.9 or non-Grbl controllers are not supported by the sender.

Creat-new, 2023-04-29:
I think it's my problem, I'm very sorry. Based on your prompt, I have resolved the problem. Thank you again @phil-barrett @terjeio

closed 2023-04-29 by terjeio