terjeio/ioSender #292: Settings Guide

dontfeelcold, 2023-05-02:
In the version 2.0.40 under "Settings: Grbl", there used to be text under the tick boxes explaining what the various settings did. In the current version 2.0.42, that text has disappeared.

It would be nice to have it back.

Also, each time I go between each setting, there is a lag in the page loading.

Neither problem exist in 2.0.40

dontfeelcold, 2023-05-02:
version 2.0.40
%
; grblHAL
; 1.1f.20210715
; [OPT:VNMSL+,999,1024,4,0]
; [NEWOPT:ENUMS,RT+,ES,TC,ETH]
; [FIRMWARE:grblHAL]
; [NVS STORAGE:*FLASH]
; [DRIVER:iMXRT1062]
; [DRIVER VERSION:210725]
; [DRIVER OPTIONS:USB.2]
; [BOARD:T41U5XBB]
; [AUX IO:1,3,0,0]
; [NETCON:Telnet]
; [IP:192.168.5.1]
; [PLUGIN:ENCODER v0.01]
;
$N0=
$N1=
; 0 - Step pulse time
$0=10.0
; 1 - Step idle delay
$1=25
; 2 - Step pulse invert
$2=0
; 3 - Step direction invert
$3=1
; 4 - Invert step enable pin(s)
$4=0
; 5 - Invert limit pins
$5=0
; 6 - Invert probe pin
$6=0
; 7 - Disable spindle with zero speed
$7=0
; 10 - Status report options
$10=511
; 11 - Junction deviation
$11=0.010
; 12 - Arc tolerance
$12=0.002
; 13 - Report in inches
$13=0
; 14 - Invert control pins
$14=78
; 15 - Invert coolant pins
$15=0
; 16 - Invert spindle signals
$16=0
; 17 - Pullup disable control pins
$17=0
; 18 - Pullup disable limit pins
$18=0
; 19 - Pullup disable probe pin
$19=0
; 20 - Soft limits enable
$20=0
; 21 - Hard limits enable
$21=0
; 22 - Homing cycle
$22=0
; 23 - Homing direction invert
$23=0
; 24 - Homing locate feed rate
$24=25.0
; 25 - Homing search seek rate
$25=500.0
; 26 - Homing switch debounce delay
$26=250
; 27 - Homing switch pull-off distance
[... 373 lines cut ...]
; 343 - Tool change locate feed rate
$343=25.0
; 344 - Tool change search seek rate
$344=200.0
; 345 - Tool change probe pull-off rate
$345=100.0
; 347 - Dual axis length fail
$347=5.0
; 348 - Dual axis length fail min
$348=2.500
; 349 - Dual axis length fail max
$349=25.000
; 370 - Invert I/O Port inputs
$370=0
; 372 - Invert I/O Port outputs
$372=0
; 400 - Encoder mode
$400=0
; 401 - Encoder counts per revolution
$401=400
; 402 - Encoder counts per detent
$402=4
; 403 - Encoder double click sensitivity
$403=500
%

terjeio, 2023-05-02:
Ok, thats an old controller version. You are either stuck with 2.0.40 or you have to update the controller firmware. You may use the [Web Builder](http://svn.io-engineering.com:8080/) to create a binary if you decide to update.

dontfeelcold, 2023-05-02:
Fixed