terjeio/ioSender #293: Feature request: spindle_up and spindle_down time delay.

RDS356, 2023-05-03:
In my though this feature must be on sender(like mach3, Bcnc, etc) and not in firmware.
I also though that the working is simple;
When M3 or M4 enable respective pin and wait spindle_up in ms for executing gcode or joging.
When M5, disable and wait spindle_down in ms too.

phil-barrett, 2023-05-03:
The current "state of the art" is to have CAM insert a dwell (G4) in tool path (aka Post) processing.  The (obvious) disadvantage to that is you need to put it in every CAM you use.  Having it in a sender removes that though becomes an issue if you use different senders.  I do find myself hitting feed hold to accommodate spin up more than I would like - over the course of a year, I probably use 10 different CAMs.

On the other hand, there is a reasonable argument for having a dwell setting in the machine controller. Since it is for a specific machine (and more specifically the spindle in that machine), a spin up/down dwell should be part of the basic machine operation. If I was a CNC machine manufacturer, I'd want that as a differentiator.

RDS356, 2023-05-04:
Well, if this feature is available only in the sender is much more easy to control. See, we have a lot of CAM(I never saw this feature in any CAM), and just one sender(mostly). So we do not need have to worry with post processing or other.
I never saw this feature in any VFD too.

Do you agree ?

phil-barrett, 2023-05-04:
I agree that in CAM is not ideal but it is currently the only game in town.  Most Post Processor "languages" allow you to insert GCode into at least the spindle-on sequence.  I added G4 P5 right after the M3 in my Vectric one for grblHAL, for example.  

I agree with the point about the VFD having it - that's the most logical place for it.  I wouldn't hold my breath, though.

Yeah, putting it in the Sender is pragmatic. On that point we agree.

terjeio, 2023-05-04:
IMO this should be handled in the controller - via the _spindle at speed_ functionality in the core.
There are several ways this can be implemented:
* input from spindle encoder.
* input from Modbus controlled VFD.
* input from end of ramped spin up.
* configurable delay.

The first three are already available depending on the driver/build configuration. The first two by setting $340 (in the _Spindle_ settings group) to a non-zero value when the setting is available. Ramped spin up is currently available for a few [drivers](https://github.com/grblHAL/drivers) - more should be upgraded for that...
A configurable delay could be implemented as a plugin.

phil-barrett, 2023-05-04:
All the more reason to move to ModBus. Though, there are lots of machines without it so a configurable delay makes a lot of sense.