terjeio/ioSender #296: Spindle PWM frequency does not match settings

300bar, 2023-05-25:
Hi, I am trying to implement a spindle PWM control using ioSender (2.0.42) and a T41U5XBB board with a grblHal downloaded from Terje´s web builder. My problem is that after spending a bunch of time, can not see why the output PWM frequency does not match the frequency set by ioSender grbl settings, if I try 50Hz, I get more than 1 KHz, changing the frequency settings does not have any correlation with the output frequency values.
Any help would be much appreciated.

closed 2023-05-25 by terjeio

terjeio, 2023-05-25:
FYI I have now implemented low frequency PWM for the iMXRT1062, will likely commit an update in the next few days when additional PWM outputs are tested and ready.

300bar, 2023-05-26:
Hi Terje, thank you! much appreciated.
I will wait for the update.

terjeio, 2023-05-27:
Up now, if still issues report for the [iMXRT1062 driver repo](https://github.com/grblHAL/iMXRT1062).