terjeio/ioSender #298: ALARM1

tcurdt, 2023-05-29:
I just had a very weird episode with ioSender.

I've updated settings to see if hard limits are the way to go.
After the change I was getting an "ALARM1".
And the error description doesn't really make sense to in the current context.

There wasn't much I can do. Unlock didn't work. And restart also seemed to have issues.
So I went back into the settings to change it back. But the settings were just a white screen.

After a lot of restarting and poking I somehow got access to the settings again and changed it back.
It seems to be working again now.

I don't quite understand if this was a firmware problem, or whether it was ioSender having a bad day.
I'll try to gather more details with the latest build - but if you already got some input I am all ears.

terjeio, 2023-05-30:
> I don't quite understand if this was a firmware problem

It is a firmware problem carried over from legacy Grbl: alarm 1 is an event/state that block all commands until a reset and/or the condition is cleared (depends on a setting). FYI I recently [updated grblHAL](https://github.com/grblHAL/core/blob/master/changelog.md#20230429) to allow $-commands when in a "blocking" state - this, and later, versions will behave better in such circumstances.

tcurdt, 2023-05-30:
I upgraded to ioSender edge and switching to hard limits and returning from ALARM1 seems to work fine now.
Thanks for the help!

closed 2023-05-30 by tcurdt