terjeio/ioSender #299: how to resolve missing G64 in IOsender and Grblahal?

evangois, 2023-06-06:
I'm using iosender on a tangential cutting machine, where the cutting knife uses the A axis to direct the knife. I use a postprocessor that generates interpolation with G1 for x and y, it is working but the movement is shaky, it would be necessary to use G64 but observe that grblhal does not understand this instruction. Would there be any way to do this, to leave the movement continuous and smooth.?

andrewmarles, 2023-06-06:
You can use the junction deviation to increase the radius that is used to manage corner acceleration, this is basically how G64 works (with the P parameter).  If you set it to an arbitrarily small value you will have basically exact stop mode (G61).

If you are using linear interpolation you may need to increase your block buffer, depends on the size of your segments and which controller you are using.

terjeio, 2023-06-06:
$398 (Planner buffer blocks) controls the block buffer, 35 is the default value. How much it can be increased depends on the controller (free memory). 100 or more will likely help, 1000 is max. If set to a too high value it will be adjusted down automatically to fit available memory.
Also check that _Agressive buffering_ is turned on in the sender, you will find that option in the _Settings: App_ tab.

terjeio, 2023-12-30:
G64 support is beeing discussed/worked on - ref. [this discussion](https://github.com/grblHAL/core/discussions/412).

Closing this issue.

closed 2023-12-30 by terjeio