terjeio/ioSender #301: No support for GRBL on Arduino Nano?

Dexus, 2023-06-15:
I have a china clone of an Arduino Nano with CH3xx but I'm not able to connect to it.

terjeio, 2023-06-15:
Odd, are you sure that no other program has the connection open?

terjeio, 2023-06-15:
Delete App.config and try again, same issue?

terjeio, 2023-06-15:
[This question](https://stackoverflow.com/questions/76158517/serial-to-usb-cable-ch340-new-driver-not-functioning-with-c-sharp-applicatio) could be a clue? It seems that I do not have any CH340 based boards to test with so I am kind of stuck when it comes to debugging this. My Mega and Due are both genuine boards with Atmel USB<>UART chips and they work fine, and the CP210x based ESP32 board does as well.

Dexus, 2023-06-20:
Hmm, looks like that could be a clue. But wired that it works with other programs fine... only ioSender does not work currently.

closed 2023-06-20 by Dexus