terjeio/ioSender #310: SD Card, Download and Run, no Program tab or Run time update,

phil-barrett, 2023-07-09:
ioSender 2.0.42  

In SD Card panel, right click on valid GCode file,  select Download and Run
It switches back to Grbl panel, the program runs, 3D View updates correctly but Program stays on first block and Run time in lower right stays at 0:00:00

If you use Run (SD Card panel, right click, select Run), Program panel remains blank but Run time in lower right does update correctly.

Pico grblHAL, 
[VER:1.1f.20230704:]
[OPT:VNMSL2,35,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,ES,SED,RTC,WIFI,FTP,SD,YM]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:RP2040]
[DRIVER VERSION:230525]
[DRIVER OPTIONS:SDK_1.5.0]
[BOARD:PicoCNC]
[AUX IO:1,8,0,0]
[WIFI MAC:28:cd:c1:06:dd:2a]
[IP:192.168.0.101]
[NETCON:Telnet]
[PLUGIN:SDCARD v1.09]
[PLUGIN:FS macro plugin v0.03]

phil-barrett, 2023-07-09:
update.  It seems that the previous value of Run Time is left over when you do a Download and run.

terjeio, 2023-07-09:
I'll fix the Run Time issue. However, keeping the program listing in sync is not easy to get working - it would likely require gcode with line numbers and line number reporting enabled. Or streaming the gcode from the sender instead of from the SD card...

phil-barrett, 2023-07-09:
OK.  Thanks for looking into it.

terjeio, 2023-07-11:
The latest [edge version](http://www.io-engineering.com/downloads) has a fix for the Run Time issue.