terjeio/ioSender #311: Starting ioSender 2.0.42 stuck at "MPG mode active"

malcolmmp, 2023-07-10:
I recently updated my grblHAL firmware using the online firmware builder, and ioSender was initially working with that. I reloaded my configuration, used my laser cutter, and then it sat for a little while.

Fast forward to today, and I can't bring up ioSender. I'm simply given a dialog saying "MPG mode active, waiting for it to be disabled..."

Running an older version of ioSender (2.0.37) just throws an unhandled exception. Attached are screenshots of each.

[image]
[image]

grblHAL firmware is at 20230626, and nothing should have been included with regards to MPG support when I 

I primarily use LaserWeb4 and I'm able to connect and operate normally there, however calibration and troubleshooting ops are much easier with ioSender.

Are there any logs or settings I can dig around in to begin to sort out this issue? Or simply a command I can issue from the LW4 console to disable MPG mode and make ioSender happy?

Thanks
Malcolm

terjeio, 2023-07-11:
`$481` (auto report interval) is > 0?

malcolmmp, 2023-07-11:
Thanks @terjeio. I'll take a look tomorrow when I'm in the shop and report back.

I certainly could have made a change to that setting as I was attempting to get the "DRO" in LW4 working.

malcolmmp, 2023-07-11:
That did the trick!

I think I must have seen the auto report and turned it on in hopes that LaserWeb4 would parse it and display the realtime position (it didn't).

I didn't see anything for documentation regarding the $481 setting in the grblHAL or ioSender wikis, but I might not have been looking in all the right places.

If you'd find it useful, I'd be happy to submit a couple of PRs in the docs around the setting and the issue I ran into.

Thank you so much @terjeio, for both the excellent project and fast helpful response!

closed 2023-07-11 by malcolmmp

terjeio, 2023-07-21:
> I didn't see anything for documentation regarding the $481 setting in the grblHAL or ioSender wikis, but I might not have been looking in all the right places.

I have not documented this since it is a new controller feature and I have not updated the sender for it. I just uploaded an [edge version](http://www.io-engineering.com/downloads) that should detect if auto reporting is enabled and if so disable it for the session. So I believe adding documentation for this is not needed - thanks for the offer though.