terjeio/ioSender #313: Establish reference offset - probe fail

rmaltais, 2023-07-11:
When i tried to setting reference, when message appear for assert probe, on second start nothing do. A message in status show probe fail. I can't see the verbose console because le tab is ot enable. I can't do anything else.
If i bypass the assert probe, the first touch move then on second move, the probe touch and fail message again in status. The probe didn't want to move and i can't go on control again. In this case, i have no choice and must close prgram.

Sorry my poor english, i'm canadian french. Hope you understand this poblem. Use a  GRBLHAL STM32F1XX driver.
My probe is NO and close on GND.

terjeio, 2023-07-13:
What is your $10 setting? _Probe coordinates_ must be added to the status report but I can't see them in the log.

[image]

rmaltais, 2023-07-13:
Hi,

I don't know is the reason of my problem. After activate probe coordinate like you explain, my probe work perfecfectly now. if is the reason because, thank for your help. All work fie now!

rmaltais, 2023-07-13:
Thank for help

closed 2023-07-13 by rmaltais