terjeio/ioSender #318: Probing for Center 2.0.42

pierrerdurand, 2023-07-21:
In the preview window probing for center, I see an exclamation mark (!) before 4 G1 instructions:
When executing the probe command it pauses at the end of the previous line: G38.3F80X-10
What am I doing wrong?

G1F2000Z-5
G1F2000X-6.075
G38.3F80X-10
!G1F2000X0.5
G38.3F20X-2
G1F2000X6.075
G38.3F80X10
!G1F2000X-0.5
G38.3F20X2
G1F2000X0
G1F2000Y-6.075
G38.3F80Y-10
!G1F2000Y0.5
G38.3F20Y-2
G1F2000Y6.075
G38.3F80Y10
!G1F2000Y-0.5
G38.3F20Y2
G1F2000Y0
G1F2000Z0
; Post XY probe
G1F2000X0Y0
G10L2P1X0Y0

[image]

terjeio, 2023-07-21:
>  I see an exclamation mark (!) before 4 G1 instructions

This is used internally by the probing executor to discard the previous result, it is not sent to the controller. 

> When executing the probe command it pauses at the end of the previous line

What do you mean by "pauses"? A short delay before continuing? Entering feed hold?
Enable _Verbose_ mode in the Console  and clear it then run again. Post the console log in a comment.

You may also try the latest [edge version](http://www.io-engineering.com/downloads) which is the one I use for testing.

pierrerdurand, 2023-07-21:
When I say pause, I mean that as soon as the probe makes the first touch on the inner circle it stops moving without terminating the probing (see screen two) where the only option is to click on the Stop button, which I did press.
I attach two screenshots: one before the start and the other when it pauses.
I also copy the content of the console from the launch of IOSender until after I pressed the Stop button.
[image]
[image]

This is a new Teensy 4.1 GRBLHal controller with firmware built with the Web Builder three days ago.

pierrerdurand, 2023-07-21:
This is the content of the Console:

GrblHAL 1.1f ['$' or '$HELP' for help]
[VER:1.1f.20230718:]
[OPT:VNMZHTSL2,35,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,ES,REBOOT,TC,SED,RTC,ETH]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:iMXRT1062]
[DRIVER VERSION:230604]
[DRIVER OPTIONS:USB.1]
[BOARD:T41U5XBB]
[AUX IO:4,3,0,0]
[IP:10.0.0.24]
[NETCON:Telnet]
[G54:-272.811,264.115,-121.702]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:-187.000,61.000,-116.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[HOME:0.000,0.000,0.000:7]
[TLO:0.000,0.000,0.000]
[PRB:-197.818,61.004,-115.998:1]
[G54:-272.811,264.115,-121.702]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:-187.000,61.000,-116.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[HOME:0.000,0.000,0.000:7]
[TLO:0.000,0.000,0.000]
[PRB:-197.818,61.004,-115.998:1]
<Idle|MPos:-224.553,144.884,-109.999|Bf:35,1023|FS:0,0|WCO:-272.811,264.115,-121.702|WCS:G54|A:|Sc:|H:1,7|T:0|TLR:0|FW:grblHAL>
[GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]
ok
[G54:-272.811,264.115,-121.702]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:-187.000,61.000,-116.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[HOME:0.000,0.000,0.000:7]
[TLO:0.000,0.000,0.000]
[PRB:-197.818,61.004,-115.998:1]
ok
<Idle|MPos:-224.553,144.884,-109.999|Bf:35,1023|FS:0,0|WCO:-272.811,264.115,-121.702>
<Idle|MPos:-224.553,144.884,-109.999|Bf:35,1023|FS:0,0|WCO:-272.811,264.115,-121.702|WCS:G54|A:|Sc:|H:1,7|T:0|TLR:0|FW:grblHAL>
Probing, pass 1 of 2...
G91F80
ok
PM:ok
G53G1F200Z-114.999
ok
PM:ok
G53G1F200X-230.378
ok
PM:ok
G38.3F80X-15
Init failed! 
<Run|MPos:-224.553,144.884,-110.130|Bf:33,1023|FS:240,0|Pn:S|WCO:-272.811,264.115,-121.702|WCS:G54|A:|Sc:|H:1,7|T:0|TLR:0|FW:grblHAL>
<Run|MPos:-224.553,144.884,-110.140|Bf:33,1023|FS:252,0|WCO:-272.811,264.115,-121.702|WCS:G54|A:|Sc:|H:1,7|T:0|TLR:0|FW:grblHAL>
<Run|MPos:-227.320,144.884,-114.998|Bf:34,1023|FS:400,0|WCO:-272.811,264.115,-121.702|WCS:G54|A:|Sc:|H:1,7|T:0|TLR:0|FW:grblHAL>
<Run|MPos:-228.745,144.884,-114.998|Bf:34,1023|FS:400,0>
<Run|MPos:-230.017,144.884,-114.998|Bf:34,1023|FS:126,0>
Init failed! 
<Run|MPos:-230.401,144.884,-114.998|Bf:34,1023|FS:80,0|Pn:S>
<Run|MPos:-230.411,144.884,-114.998|Bf:34,1023|FS:80,0|WCO:-272.811,264.115,-121.702|WCS:G54|A:|Sc:|H:1,7|T:0|TLR:0|FW:grblHAL>
<Run|MPos:-232.098,144.884,-114.998|Bf:34,1023|FS:80,0|WCO:-272.811,264.115,-121.702|WCS:G54|A:|Sc:|H:1,7|T:0|TLR:0|FW:grblHAL>
<Run|MPos:-232.370,144.884,-114.998|Bf:34,1023|FS:80,0|Ov:200,100,90>
<Run|MPos:-232.653,144.884,-114.998|Bf:34,1023|FS:80,0>
<Run|MPos:-232.916,144.884,-114.998|Bf:34,1023|FS:80,0>
<Run|MPos:-233.189,144.884,-114.998|Bf:34,1023|FS:80,0>
<Run|MPos:-233.451,144.884,-114.998|Bf:34,1023|FS:80,0>
<Run|MPos:-233.724,144.884,-114.998|Bf:34,1023|FS:80,0>
<Run|MPos:-234.007,144.884,-114.998|Bf:34,1023|FS:80,0>
<Run|MPos:-234.279,144.884,-114.998|Bf:34,1023|FS:80,0>
<Run|MPos:-234.542,144.884,-114.998|Bf:34,1023|FS:80,0>
<Run|MPos:-234.815,144.884,-114.998|Bf:34,1023|FS:0,0|Pn:P>
[PRB:-234.784,144.884,-114.998:1]
ok
PM:ok
<Idle|MPos:-234.926,144.884,-114.998|Bf:35,1023|FS:0,0|Pn:P>
<Idle|MPos:-234.926,144.884,-114.998|Bf:35,1023|FS:0,0|Pn:P|WCO:-272.811,264.115,-121.702>
<Idle|MPos:-234.926,144.884,-114.998|Bf:35,1023|FS:0,0|Pn:P|Ov:200,100,90>
PM:cancel
G90
Probing cancelled/failed.
ok
<Idle|MPos:-234.926,144.884,-114.998|Bf:35,1023|FS:0,0|Pn:P|WCO:-272.811,264.115,-121.702|WCS:G54|A:|Sc:|H:1,7|T:0|TLR:0|FW:grblHAL>
Probing failed, probe signal is asserted.
Init failed! Probing failed, probe signal is asserted.
<Idle|MPos:-234.926,144.884,-114.998|Bf:35,1023|FS:0,0|Pn:P|WCO:-272.811,264.115,-121.702>
G90
G90
ok
<Idle|MPos:-234.926,144.884,-114.998|Bf:35,1023|FS:0,0|Pn:P|WCO:-272.811,264.115,-121.702|WCS:G54|A:|Sc:|H:1,7|T:0|TLR:0|FW:grblHAL>
<Idle|MPos:-234.926,144.884,-114.998|Bf:35,1023|FS:0,0|Pn:P|WCO:-272.811,264.115,-121.702>

pierrerdurand, 2023-07-24:
To ensure that the build of the board was correct, I loaded a prebuilt firmware (deprecated) from the grblhal.org web site and the probing works correctly.

phil-barrett, 2023-07-25:
I highly recommend you use the [grblHAL Web Builder](http://svn.io-engineering.com:8080/) for much more up to date firmware.

terjeio, 2023-07-25:
I have not been able to replicate this behaviour with my test machine (MSP432 based).
What is bit odd is that your log contains "Init failed" messages while the probing sequence is running, this can happen if
1. Cycle start input is triggered. I see that this happened at least once.
2. \<Alt>+\<R\> is pressed on the keyboard.
3. The _Start_ button is pressed. This should be disabled so not possible.

I have added code blocking 1 and 2 from triggering the init function while a sequence is running, perhaps this will help. I have just uploaded new [edge versions](http://www.io-engineering.com/downloads) for you to try.

---

Do you have a switch connected to the cycle start input? And did you press it during probing?
If not then EMI is somehow triggering the input even if the T41U5XBB board should be fairly immune due to having optocouplers on all inputs.

pierrerdurand, 2023-07-25:
I will rebuild the firmware, test with and without the cycle start switch and review the protection against EMI.
I attach a photo of the circuit board in my enclosure and if anyone sees a weakness they are welcome to critique.
[image]

terjeio, 2023-07-26:
Nice cabinet!
Before you start tinkering with the controller please try with an [edge version](http://www.io-engineering.com/downloads) of the sender.

pierrerdurand, 2023-07-26:
Willdo Terje…

P

pierrerdurand, 2023-07-26:
I have loaded the ioSender Edge version and the latest firmware for grblHal using the Web Builder and the probing works fine.
Have done multiple tests successfully.
Thank you very much Terje.

closed 2023-08-26 by terjeio