terjeio/ioSender #319: RP2040 & ioSender MDI G Code completion

phil-barrett, 2023-07-28:
ioSender 2.0.42, PR2040 recent build. I think this is an ioSender problem but could be RP2040 related.

Initially after starting ioSender XL, I can issue GCode commands via MDI with no problem.  If I load a GCode program, G0 commands issued via MDI do not terminate.  The expected behavior happens (G0X0 goes to zero, for example).  However, the tabs at the top, except for Grbl, are greyed out and the Run time in the lower right continues to run. I can hit Feed Hold but Run time continues to advance.  If I press stop, Run time stops accumulating and the tabs at the top become ungreyed (not sure that's a real word).  I can also press the red Reset button and stop it.  I verified that grblHAL is still running on the controller (killed ioSender, used terminal app to talk to the controller, was able issue gcode commands, look at registers and so on.). 

Further weirdness.  After loading a GCode Program.  The jog buttons misbehave.  You can press any X, Y or Z button once.  It works but subsequent jog button presses do nothing.  Prior to loading a GCode Program, they behave normally.

GCode Programs seem to execute correctly.

Perhaps a clue - macros seem to execute correctly.  They terminate properly. The same GCode entered into the MDI does not.

This does not happen with ioSender and the iMXRT1062 port.  

Clipboard data.
```
%
; grblHAL
; 1.1f.20230714
; [OPT:VNMSL2,35,1024,3,0]
; [NEWOPT:ENUMS,RT+,ES,SED,RTC,SD]
; [FIRMWARE:grblHAL]
; [NVS STORAGE:*FLASH]
; [DRIVER:RP2040]
; [DRIVER VERSION:230525]
; [DRIVER OPTIONS:SDK_1.5.0]
; [BOARD:PicoCNC]
; [AUX IO:1,8,0,0]
; [PLUGIN:MODBUS v0.14]
; [PLUGIN:SDCARD v1.10]
; [PLUGIN:FS macro plugin v0.04]
;
$N0=
$N1=
; 0 - Step pulse time
$0=10.0
; 1 - Step idle delay
$1=25
; 2 - Step pulse invert
$2=0
; 3 - Step direction invert
$3=0
; 4 - Invert stepper enable pin(s)
$4=7
; 5 - Invert limit pins
$5=7
; 6 - Invert probe pin
$6=0
; 8 - Ganged axes direction invert
$8=0
; 9 - PWM Spindle
$9=1
; 10 - Status report options
$10=511
; 11 - Junction deviation
$11=0.010
; 12 - Arc tolerance
$12=0.002
; 13 - Report in inches
$13=0
; 14 - Invert control pins
$14=70
; 15 - Invert coolant pins
$15=0
; 16 - Invert spindle signals
$16=4
; 17 - Pullup disable control pins
$17=0
; 18 - Pullup disable limit pins
$18=0
; 19 - Pullup disable probe pin
$19=0
; 20 - Soft limits enable
$20=0
; 21 - Hard limits enable
$21=0
; 22 - Homing cycle
$22=0
; 23 - Homing direction invert
$23=0
; 24 - Homing locate feed rate
$24=25.0
; 25 - Homing search seek rate
$25=500.0
; 26 - Homing switch debounce delay
$26=250
; 27 - Homing switch pull-off distance
$27=1.000
; 28 - G73 Retract distance
$28=0.100
; 29 - Pulse delay
$29=5.0
; 30 - Maximum spindle speed
$30=1000.000
; 31 - Minimum spindle speed
$31=0.000
; 32 - Mode of operation
$32=0
; 33 - Spindle PWM frequency
$33=5000
; 34 - Spindle PWM off value
$34=0.0
; 35 - Spindle PWM min value
$35=0.0
; 36 - Spindle PWM max value
$36=100.0
; 37 - Steppers deenergize
$37=0
; 39 - Enable legacy RT commands
$39=1
; 40 - Limit jog commands
$40=0
; 43 - Homing passes
$43=1
; 44 - Axes homing, first pass
$44=4
; 45 - Axes homing, second pass
$45=3
; 46 - Axes homing, third pass
$46=0
; 62 - Sleep enable
$62=0
; 63 - Feed hold actions
$63=2
; 64 - Force init alarm
$64=0
; 65 - Probing feed override
$65=0
; 100 - X-axis travel resolution
$100=250.000
; 101 - Y-axis travel resolution
$101=250.000
; 102 - Z-axis travel resolution
$102=250.000
; 110 - X-axis maximum rate
$110=5000.000
; 111 - Y-axis maximum rate
$111=5000.000
; 112 - Z-axis maximum rate
$112=5000.000
; 120 - X-axis acceleration
$120=1000.000
; 121 - Y-axis acceleration
$121=1000.000
; 122 - Z-axis acceleration
$122=1000.000
; 130 - X-axis maximum travel
$130=200.000
; 131 - Y-axis maximum travel
$131=200.000
; 132 - Z-axis maximum travel
$132=200.000
; 341 - Tool change mode
$341=0
; 342 - Tool change probing distance
$342=30.0
; 343 - Tool change locate feed rate
$343=25.0
; 344 - Tool change search seek rate
$344=200.0
; 345 - Tool change probe pull-off rate
$345=200.0
; 346 - Restore position after M6
$346=1
; 370 - Invert I/O Port inputs
$370=0
; 372 - Invert I/O Port outputs
$372=0
; 374 - ModBus baud rate
$374=3
; 375 - ModBus RX timeout
$375=50
; 384 - Disable G92 persistence
$384=0
; 398 - Planner buffer blocks
$398=35
; 481 - Autoreport interval
$481=0
%
```

phil-barrett, 2023-07-28:
Further testing.  
ioSender 2.0.42 (non XL version) has the same issue.
ioSender XL 2.0.40 has the same issue.
Candle GCode Sender does not have the issue.
Open Builds Control does not have the issue.

terjeio, 2023-07-28:
This is due to a recent change I made in cycle start signal reporting that has unintended side-effects. ioSender monitors the cycle start signal in the real-time report and may use that to start a loaded program or a probing sequence, this to avoid reaching for the keyboard or the mouse to do so. The issue I had with this is that sometimes cycle start is not reported if the keypress (or command from an MPG/keypad) is shorter in duration than the reporting interval.

I believe I have found a better solution for it, I will test a bit more and commit a fix tomorrow.

phil-barrett, 2023-07-28:
Thanks for looking at it so quickly!

terjeio, 2023-07-29:
[Fix](https://github.com/grblHAL/core/commit/280dd953c3aaa9edd79cd14270fbcf6e717f8518) committed.

phil-barrett, 2023-07-29:
Tested.  Looks like that fixed it.  Thanks for the fast response.  You rock!

closed 2023-08-26 by terjeio