terjeio/ioSender #323: Big impossible feature request

crackinx, 2023-08-25:
Might not even be an iosender thing or a grblhal thing, but I was thinking would it be possible to be able to hook up a 4th rotary axis as it's own stand alone device instead of needing a controller with another available axis? Either linking it with grblhal wit uart or i2c, or having iosender connect to 2 serial devices to control the rotary axis independently. Might be a huge under taking, but would be worth it to me to just be able to plug in the stand alone rotary axis.

terjeio, 2023-08-25:
With synchronized motion? Forget it, even if you paid me I would not do that.
A 4+ axis controller is not expensive, cheapest is around USD 15...

crackinx, 2023-08-25:
That's fair, was just an idea, mostly because I have everything I would need to build it laying around without needing to spend money and re-build my controller box with a new board.

closed 2023-09-21 by terjeio