terjeio/ioSender #324: Windows: Unable to select new COM port if prior selected port has invalid / null response / timeout

DrFrankReade, 2023-08-25:
Reproduce: Start a fresh install of IOSender. Select the COM port that your UPS / Arduino or whatever is on.  Continue, ioSender will exit without allowing selecting an alternate COM port. Subsequent restarts will not allow selection of a new, valid COM port.

[image]

Work-around is to delete App.config and restart.

phil-barrett, 2023-08-25:
You can get ioSender to put up the dialog by unplugging the USB port, restarting ioSender and after the dialog comes up, plugging USB back in.  You can also specify PORT as a command line option. I suggest this for changing the com mechanism from USB to Ethernet or WiFi.

But, there must be something else going on prevent ioSender and grblHAL communicating.

terjeio, 2023-08-26:
You have three options:

1. Delete the _app.config_ file in the sender folder and restart as you have discovered.
2. Set the port on the command line in a shortcut.
3. Force the sender to show the port selection dialog on startup, again by a command line parameter. The previous port selection is remembered.

See [this wiki page](https://github.com/terjeio/ioSender/wiki/Setup-and-configuration) for details.