terjeio/ioSender #334: ioSender.XL?

gharbarka, 2023-10-04:
I would like to install ioSender on my PC. There are 2 versions in the .zip list with one on them being referred to as XL.

I have scanned the past issues titles as well as the wiki but I could not find any reference.

Can somebody please explain the differences?

phil-barrett, 2023-10-04:
ioSender:

phil-barrett, 2023-10-04:
ioSender. you have to open the jog panel to jog with the mouse.
[image]

ioSender XL gets rid of the jog panel and has the controls always present on the main panel.
[image]

gharbarka, 2023-10-04:
Gosh...that fast. Thank you.

closed 2023-10-04 by gharbarka