terjeio/ioSender #339: ioSender v2.0.42 Many project files are missing at compile time.

ShenRen, 2023-11-09:
[image]

CNC Controls
Unable to find file ”Themes\BlackTheme.xaml"
Unable to find file ”Themes DarkTheme.xaml"
Unable to find file “ThemeslWhiteThemexaml"
Unable to find file ”ThemeslStandardTheme.xaml”
Unable to find file ”Themes\Dictionary.xaml”
Unable to find file ”Themes\LightTheme.xaml”

ioSender
Unable to find file ”..CNC Controls\CNC ControlsThemes ThemeSelector.xaml”

terjeio, 2023-11-12:
I believe these are the missing files:
[Themes.zip](https://github.com/terjeio/ioSender/files/13327494/Themes.zip)

I'll consider adding them in the next commit, currently they are not in use and might be replaced by something else for theming.

ShenRen, 2023-11-13:
Glad to hear from you. 

I have added the files to the project, but several files are still missing. 

```
Themes\Dictionary.xaml
Themes\ThemeColors.cs
```

I'm not sure if there will be more compilation errors about missing files. If you can continue to give me the file, I will continue to compile.

I've rolled the project back to v2.0.42 and it compiles correctly.

Thanks for the whole grblhal project, it's really awesome!