terjeio/ioSender #343: Understandings and .ini Files

silver2row, 2023-12-04:
Hello,

I am researching the .ini files needed for a cartesian laser machine and iosender w/ grblHAL along w/ Barrett's Board.

Where would I place and use the ini file? What would be in the ini file?

Seth

silver2row, 2023-12-04:
Hello...thank you. So, when I set `$130 X458` and `$131 Y458`, it sets my boundaries of how far I can travel...okay.

Also, I need to set Feed Rate at some interval. Okay. `F500` will accomplish this idea. Okay...

I cut out homing for now b/c of my location issues on the cartesian plane. It seems my Y_Axis keeps stopping short of the homing end stop switch which is the min switch. The X_Axis min end stop works but b/c the Y_Axis end stop stops prematurely, the X_Axis never homes, i.e. hence me cutting out homing for now.

Seth

P.S. What is a normal feed rate for a cheap machine? I have good drivers but my motors are silly and my machine motors can handle the frame easily. The frame is cheap. I have been working w/ this cheap machine to get better acquainted w/ grblHAL and iosender for use on Windows 11 machines.

silver2row, 2023-12-04:
Also sir,

I do this...

```
1. $130 X458
2. $131 Y458
3. G0 Y25 F25
```

After number 3. has been applied, the machine goes on forever and not just 25mm at a feed rate of 25.

Is this due to not having homing set up and the limit switches being set up?

Seth

silver2row, 2023-12-04:
# Update

I read one of your articles on your website, sir. Nice! Imperial was used on the software for CAD/CAM while the iosender source used metric. Blah. Okay. I will figure this out. Sorry for harassing you.

Seth

terjeio, 2023-12-04:
> P.S. What is a normal feed rate for a cheap machine?

Impossible to tell, acceleration plays a role as well the hardness of the material you want to cut, depth of cut and the size of the tool... If not cutting you may try increasing the both the max feed rate and the acceleration until the motor stalls and back off from the values where stalling occurs. 

> Is this due to not having homing set up and the limit switches being set up?

Not really. G0Y25 commands movement to the _absolute_ position 25mm, how large the actual movement will be depends on the position the movement starts from. If the starting position is not 0 it can be smaller or larger.

Have you [calibrated your axes](https://duckduckgo.com/?t=ffab&q=grbl+calibrate+motors&ia=web) so that steps/mm matches the machine? Most steppers produces one turn per 200 steps. E.g. if you have stepper drivers set to 8 microsteps and a screw that produces 2mm movement per turn then the steps/mm setting should be 200 * 8 / 2 = 800 steps/mm. If the machine is belt driven the math is a [bit more complicated](https://zalophusdokdo.github.io/StepperMotorsCalculator/en/index.html).

phil-barrett, 2023-12-04:
[This might help](https://www.grbl.org/single-post/calibrating-grblhal-cnc-system-movement) you calibrating.  You don't even need to work through the gearing for your belts - just use your initial guess and adjust from there.

silver2row, 2023-12-04:
Okay...thank you guys/gals. I appreciate the assistance in this endeavor. I will go to the grbl site and review things.

silver2row, 2023-12-10:
So, is it $100 X400 and not $130 X400 for stating the maximum travel the X_Axis will handle on my machine?

I see your post here but then read and reread the blog you referenced. Each seem to be different...

Seth

phil-barrett, 2023-12-10:
FWIW, you send
`$130=400`
to Grbl via MDI in your GCode sender.

silver2row, 2023-12-11:
Right. Okay, I was misreading what was typed out. You are right.

closed 2023-12-11 by silver2row