terjeio/ioSender #345: Huanyang HY01D523B 1.5kw RS-485 feedback

boboxx, 2023-12-06:
I have a HY VFD that i'm trying to configure so that I can get RPM feedback returned. I can currently start the spindle by issuing a M3 S6000 command.

My current VFD config is:
PD001=2
PD002=2
PD003=400.00
PD004=400.00
PD005=400.00
PD006=2.60
PD007=0.50
PD008=220.00
PD009=14.3
PD010=11.0
PD014=8.0
PD015=8.0
PD072=400.00
PD141=220.00
PD142=6.5
PD143=2
PD144=3000
PD163=1
PD164=2
PD165=3
PD174=7.0

My ioSender config: 
[image]

If I have ioSender $340 "Spindle at speed tolerance" set to 1.0%, my FS responce looks like this and it's not reporting.
<Idle|MPos:0.000,0.000,0.000|Bf: 35,1023|FS:0,6000,0|Ov:100,100,100|A:S>

[image]

If I dissable to $340 "Spindle at speed tolerance" set to 0.0%, then the responce looks lile this: cIdle|MPos:0.000.0.000.0.000|Bf: 35.1023|FS:0.6000,1890|SI:18.5>
[image]

What am I missing to allow my VFD to repport the current RPM so that I can use the Spindle delay?

terjeio, 2023-12-10:
This is due to a "bug" in the grblHAL ESP32 driver - will fix in the next commit, to be made available in the next few days.

boboxx, 2023-12-13:
Hmm I'm unable to select any spindle on the web builder now. Every dropdow only has a single entry called "Disabled"

terjeio, 2023-12-13:
Dropdowns works for me, can you try again? Still disabled?

boboxx, 2023-12-13:
Same problem. I tried it on 2 different computer.
[image]

boboxx, 2023-12-13:
Seems like it's working in Firefox. I had been using Safari and Edge in the past to build the FW.

terjeio, 2023-12-13:
It works for me with Edge (Version 120.0.2210.61), you have an older version or perhaps board data is somehow read from a local cache?

[image]

boboxx, 2023-12-13:
It's working from both Safari and Edge now.

By default Spindle #1 is set tp PWM

Thanks!