terjeio/ioSender #346: steps per degree: accumulating error on 4th axis

agent-r, 2023-12-07:
Hi there,

I recognised a problem when using my 4th, rotational A-axis: I did some jobs with quite a lot of rotations (spiral milling) and recognised, that after some hundreds or even thounsands of revolutions, the top face of my workpiece was turned at some degrees. I initially suspected missing steps, but since i changed for a closed-loop stepper on the A-axis recently, this was rather unlikely.

The following seems to be the problem: My axis has a 6:1 gear leading to a odd steps/degree value of 53 1/3. If i read the docs of grblHAL right, the steps/degree setting is stored as a float datatype (with about 7 decimal digits?). But in ioSender 3 decimal places (=53.333) is the most accurate value i can enter. This leads to an error of 0.12 steps per degree => 1 degree per about 444 revolutions of the axis. This machtes quite exact with the error i see on my machine.

Is there any way to set more precise values for steps/degree settings? Am I doing something wrong? Are there any (good ;-) ) reasons for the limitation to 3 decimal places? Any workarounds? Could this be improved in future releases of ioSender?

Thanks!
P

terjeio, 2023-12-08:
The sender does not limit the number of decimal places, the controller does by [telling it which format to use](https://github.com/grblHAL/core/blob/04a5edcf8353f7294167d12d32e4b073d9be9ce4/settings.c#L552).

> Is there any way to set more precise values for steps/degree settings?

Yes I believe so, try setting it via the MDI. The value will still be reported back with 3 decimal places though.

> Are there any (good ;-) ) reasons for the limitation to 3 decimal places?

Not that I know of, the limitation is inherited from legacy Grbl (and can be changed).

Note that you may also run into the limited precision of floats when commanding many revolutions of the A-axis.

agent-r, 2023-12-09:
Thanks! Setting the steps/degree value via MDI seems to work. I will check the next few days, if it really improves my parts.
I guess, from now on I will have to set this value every time after I change other settings via the ioSender grbl-settings interface, right?

terjeio, 2023-12-09:
> I guess, from now on I will have to set this value every time after I change other settings

No, only changed settings are sent to the controller.

terjeio, 2023-12-15:
FYI I [have increased the allowed number of decimals](https://github.com/grblHAL/core/blob/master/changelog.md#20231210) from 3 to 5.