terjeio/ioSender #348: Z probe with unexpected result

dragoonmc, 2023-12-14:
Trying to use a touch plate to probe Z0 WCS and always got the wrong 0 point. 

I eventually figured out the issue is that the "fixture/touch plate" thickness box changes from Fixture to touch plate based on the check box in the tool length tab. once this was unchecked Z probing worked as expected. 

My issue is that tool offset tab should not effect the probing tab like this. Is there a way to change this?

terjeio, 2023-12-14:
I am not sure what the issue is. If you are not probing a fixture at G59.3 then you should not check that check box.

> My issue is that tool offset tab should not effect the probing tab like this. Is there a way to change this?

Change what?
Remove the _Probing fixture @ G59.3 with tool:_ check box?
Not changing the _Touch plate/fixture height_ input field?
...?

dragoonmc, 2023-12-14:
I don't actually care about the check box, just that it impacts expected/desired result from a different tab's function
Can you make it so that if you aren't viewing the tool offset tab it is effectively unchecking that box? 
Either that or that check box needs to be in the other tabs as well.

terjeio, 2023-12-14:
> Can you make it so that if you aren't viewing the tool offset tab it is effectively unchecking that box?

Ok, I think I understand now - the _Touch plate/fixture height_ input field is not switched to _Touch plate_ when selecting a different tab than _Tool length offset_.
Correct? If so there might be another issue at play since the whole field including the underlying property is swapped, not only the label.

> Either that or that check box needs to be in the other tabs as well.

The underlying property is not used by the other tabs so I'll keep it as-is. What I will do is to keep track of the check box state and set/reset it depending on which tab is currently active, this since the checkbox state controls which field is visible. I'll soon make an edge build available with this change.

terjeio, 2023-12-15:
New [edge versions](http://www.io-engineering.com/downloads) are now up.

dragoonmc, 2023-12-16:
thanks, I will check it out. 
on A similar note does probing change somehow with T99 selected? I was trying to use the touch plate to do XY probing and it always tried to go through the touch plate to get back to the start location

terjeio, 2023-12-17:
T99 is not handled in any special way, other tools works ok?