terjeio/ioSender #352: Empty portlist, Windows 11, Fysetc S6, Native USB

tobjy, 2024-01-14:
When I start ioSender, no ports are listed. I have connected my Fysetc S6 board over USB. I have build and installed an GrblHAL firmware using Terjes WebBuilder, specifying Native USB for the communication. 

This is probably not an ioSender issue. I am guessing I need to install some driver for Windows 11. Can anyone out there please give me a hint?

terjeio, 2024-01-14:
This is odd. How did you flash the board?

tobjy, 2024-01-14:
I just copied the firmware.bin to a SD-card. Seems to have worked, the preinstalled Marlin firmware is gone. 

I have no experience connecting "DIY" hardware to Windows 11;  I do not know what to expect. Should a port be available in ioSender when I connect the Fysetc board, without any specific drivers installed on Windows?

terjeio, 2024-01-14:
> I just copied the firmware.bin to a SD-card.

This is the likely reason - give me a bit of time and I'll prepare a bootloader version, or maybe two, for you to try.

> Should a port be available in ioSender when I connect the Fysetc board, without any specific drivers installed on Windows?

Yes.

terjeio, 2024-01-14:
Here is one for a 32K bootloader for you to try, not 100% sure it is correct.

[firmware 32K BL.zip](https://github.com/terjeio/ioSender/files/13932898/firmware.32K.BL.zip)

There are two versions of the bootloader, you may have to [flash the latest for 0x8000](https://github.com/FYSETC/FYSETC-S6/tree/main/bootloader) first. You can check which you have by flashing Marlin - only [one of these](https://github.com/FYSETC/FYSETC-S6/tree/main/firmware/Pre-builds) will work.

You may also flash grblHAL (the Web Builder version) via DFU, but better to try the bootloader first.

tobjy, 2024-01-14:
I was able to load the Marlin 0x8000 firmware. I could see a new COM6 in the Windows device manager; ioSender also recognized the port and could start. Unfortunately I had no luck with _firmware 32K BL_. 

The booloader seems to work alright. So, I presume it would not bring anything to flash grblHAL via DFU?

terjeio, 2024-01-15:
> So, I presume it would not bring anything to flash grblHAL via DFU?

It is worth a try, with the Web Builder version. It will overwrite the bootloader though, so you will have to reflash it if you need it again.

tobjy, 2024-01-15:
You are right :-) The download worked using STM32CubeProgrammer, and I can connect with ioSender!
Many thanx!

closed 2024-01-15 by tobjy