terjeio/ioSender #353: Jogging only toggles between two positions

hallgchris, 2024-01-15:
When I try to jog using ioSender's UI, clicking a positive jog amount does nothing. A negative jog amount will alternate between moving to -1 and -2 units respectively. (sometimes it doesn't move at all too)

e.g. after subsequent presses of -X and with a jog distance of 10 mm, it goes to X=-10mm, then -20, then -10, then -20...

Here's an example of that behaviour, viewed from the console. Between jogs I changed no settings and only clicked -X

[image]

Movement by writing raw gcode works fine, and jogging is fine using other software (I tried UGS). I'm running ioSender 2.0.44 and also tried 2.0.41 and running on a different computer with no change in result.

Any ideas? Thanks :)

terjeio, 2024-01-15:
Which Grbl/GrblHAL version? (Paste output from `$I`).
Since G53 is used for jogging you have soft limits enabled the sender then tries to keep jog motion within travel limits.
Is the machine homed?
If you disable soft limits does jogging work?

hallgchris, 2024-01-16:
Here's the complete output of `$I`:
```
[VER:1.1f.20230825:]
[OPT:VNMZHS+W2,128,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,ES,EXPR,SED]
[FIRMWARE:grblHAL]
[NVS STORAGE:*EEPROM]
[FREE MEMORY:76K]
[DRIVER:STM32F446]
[DRIVER VERSION:230807]
[BOARD:FlexiHAL_20230825_111423]
[AUX IO:4,6,0,0]
[PLUGIN:MODBUS v0.14]
[PLUGIN:HUANYANG VFD v0.11]
[PLUGIN:HUANYANG P2A VFD v0.10]
[PLUGIN:Durapulse VFD GS20 v0.05]
[PLUGIN:Yalang VFD YL620A v0.02]
[PLUGIN:MODVFD v0.03]
[PLUGIN:H-100 VFD v0.02]
[PLUGIN:KEYPAD v1.3 INTERTEST]
[PLUGIN:Macro plugin v0.02]
[SPINDLE:Huanyang v1]
[PLUGIN:Probe Protection v0.01]
[PLUGIN:RGB Indicator Lights v1.0]
```

The machine is homed. Disabling soft limits did not change anything

terjeio, 2024-01-16:
Is $40 on (1)? You can find it in the _Jogging_ category in _Settings: Grbl_.
And disabling soft limits should change job commands in the console from G53 to G91, but you have to restart the sender after the change since settings are read from the controller on startup.

hallgchris, 2024-01-17:
Disabling soft limits and a restart did it - didn't realise the software had to restart. Why does soft limits cause this behaviour?

Changing $40 did not make a difference.