terjeio/ioSender #355: physical homebutton?

zurdoid123, 2024-01-18:
Heyo, amazing software. Top notch really! 

I have a question though, is it possible to add a physical button for homing? 

I'm using an skr pro 1.2 with latest version of grblhal available in the webbuilder. All working without pain and im about to add the reset, hold and cycle button. And now thought it would be very convenient if a homing button is possible aswell. No biggie if not though. Just a thought :)

terjeio, 2024-01-18:
> I have a question though, is it possible to add a physical button for homing?

Seems like it is possible since there is an aux input that might be used and there is a plugin that allows binding macros to aux inputs, enable in the _Plugins_ tab:

[image]

Documentation is currently [in the source code](https://github.com/grblHAL/Plugin_keypad/blob/6c481749765f077fe58cb95cbebd1a05debef7f7/macros.c#L23-L36), and be aware that it will not work if the pin is not interrupt capable - this is a limitation of the STM32 family of processors that sometimes arises.

Set the macro to `$H` if the setting becomes available.

zurdoid123, 2024-01-18:
Ah okey, thanks! I'll try it out.

I have another question, I have to restore the settings file every time start the software.. and I accidentally pressed the backup button thats just next to it so had to set up everything again. Am I missing something or is there no way for iosender to load it automatically?