terjeio/ioSender #356: Grbl mega 5x compatibility

Shkolik, 2024-01-23:
Hello! Thank you for you work - using ioSender on my 3 axis cnc running teensy just fine. But now I playing with 5 axis foam cutter running latest GRBL Mega 5X and having issue right from connect. Looks like ioSender parse OPT incorrectly (also reproducible on my old 3 axis cnc running grbl 1.1h). 
Controller sends OPT with value "VNMGL,35,255,64" and code try to parse 64 as NumAxes that obviously incorrect.

according to https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-v1.2-interface

> - `[OPT:]` line follows immediately after and contains character codes for compile-time options that were either enabled or disabled. 
  	- The codes are defined below and a CSV file is also provided for quick parsing. This is generally only used for diagnosing firmware bugs or compatibility issues. 
  	- The value after the first comma contains the `blockBufferSize`, as int.
  	- The value after the second comma contains the `rxBufferSize`, as int.
  	- The value after the third comma contain the `settings.flags`, as int.

so there we should parse flags, not number of axis but flags. Not sure what to do next - have no exp with grbl protocol. For now I just replaced 

`if (s.Length > 3 && NumAxes != int.Parse(s[3], CultureInfo.InvariantCulture))`

with

`if (s.Length > 3 && int.TryParse(s[3], out int flags) && NumAxes != flags && flags < 10 && flags > 0)`

to make it run, but I'm sure you know more elegant solution to this problem.

PS: Want to adapt ioSender as much as possible to FoamCutter mode, if you don't mind - digging trough code. If you interested in additional mode - let me know and I'll contribute as much as I can.

terjeio, 2024-01-23:
The correct fix is:

`if (s.Length > 3 && IsGrblHAL && NumAxes != int.Parse(s[3], CultureInfo.InvariantCulture))`

since the third element is a grblHAL extension, and also an incompatible GRBL Mega 5X extension (compared to legacy Grbl). I'll add it to the next build.

If you fork the ioSender repo and commit updates to it I can see what you are changing and may provide input on how to best proceed.

Shkolik, 2024-01-23:
Wow, didn't expect such a fast response :) Sure - will fork and do my work. Only thing - I already converted solution to .Net 4.8, but it should be a problem while it doesn't alter older C# syntax. 
Found and fixed couple minor bugs in xaml and files handling so far - maybe my changes will be useful for you. Will commit couple hours later.