terjeio/ioSender #359: After win10 updates (jan31) communicate is down (commands don't work)

MrKapim, 2024-02-03:
Hi Terje, today my laptop updated the windows (I'm use win10 pro) and after this, the ioSender don't respond to commands (close or wait for response). I tried restart the windows and tried restart the IOsender but not succcess.  Did you have informations about this problem?

MrKapim, 2024-02-03:
Oh yes important...    The ioSender "freeze" and only close if I "reset" my cnc controler (the laptop "lost" the USB communication)

phil-barrett, 2024-02-03:
Sounds strange. Check device manager to see what com ports are present - maybe there is a clue.  Also, try a different USB port.

MrKapim, 2024-02-03:
Yes, I checked the ports and is OK and tried other usb, not differences. I'm using SKR3 with STM32H743

terjeio, 2024-02-03:
Which version of ioSender?
Has the COM port driver been changed by the update? There was an issue earlier where a user had to roll back the driver,
BTW I am running Win10 Pro as well, no such porblem for me with the several boards I work with.

MrKapim, 2024-02-03:
ioSender 2.0.44 and no, there have been no driver changes. However, as I said at the beginning, ioSender opens normally and updates the information according to my firmware (it shows 4 axes, for example). But I can't do anything else because it "freezes". And it doesn't accept closing, I have to through the "windows task manager" finish, "kill" ioSender.

terjeio, 2024-02-03:
This is the driver used for the NUCLEO-F756ZG board I have:

[image]

Can you connect with a terminal program such as PuTTY and communicate with the controller?

MrKapim, 2024-02-03:
Yes. I opened the arduino and used the serial monitor to communicate and it's working. That's why I opened the conversation here...

terjeio, 2024-02-03:
I have just uploaded new [edge versions](http://www.io-engineering.com/downloads) compiled with the latest Win10 upgrades installed. It should not matter, but who knows...
The issue could be due to a .Net update gone bad? The Microsoft .Net serial port implementation has a bad rep so perhaps I should look for another option?

BTW have you tried to connect with the different _On connect_ options?

[image]

MrKapim, 2024-02-03:
Hi Terje, It's working. I use VM and reinstalled the VM tools package...
Thank you again for your strong commitment and help.

closed 2025-02-07 by terjeio