terjeio/ioSender #361: Plasma templates library

Corvuscorax45, 2024-02-13:
Hi,

is there any way to add cnc plasma cutting shape library (something similar like in attachment). Where you can only add dimensions and press start for cutting shape?
[image]
[image]

terjeio, 2024-02-13:
It is possible to add a subproject for it and make it available in a new tab - a bit similar to the [threading wizard for lathes](https://github.com/terjeio/ioSender/tree/master/CNC%20Controls%20Lathe/CNC%20Controls%20Lathe).
It is not done in an afternoon or two though - and for the time beeing I have plenty on my todo list. And I do not have a plasma cutter table to test with yet, but have started to collect bits and pieces for making one...
Do you (or anybody else) want to give it a try?

BTW is the screenshots above from a open source project?

phil-barrett, 2024-02-13:
This is something I have been thinking about for a while. There are a lot of simple operations in the shop that do not really need CAD.  Cut out a rectangle or a dado, for example. Or burn your signature onto a piece of wood.

If you (or anyone else) is going to do this, please generalize it so that you can use a router, mill or laser. For a router or mill, Feed, speed and depth of cut are useful.  For a laser, feed and power.  We are talking about a simple CAM, basically.

Making it so that shape/tool path, feed, speed, power, depth of cut can be added by the user would be super useful.

Corvuscorax45, 2024-02-14:
> It is possible to add a subproject for it and make it available in a new tab - a bit similar to the [threading wizard for [...]

As much as i would like to make it real i dont have skills to do that :) all i can i can test it on plasma because i have it or i can try to find somebody who can help to make it?

Screenshots are from chines plasma controller statfire 2100b operator manual.